Xilinx/RapidWright #164: Wire.getNode returning null

trharoldsen, 2021-04-26:
I found a wire that returns null when I try and get the associated node.

```java
Wire w = Device.getWire("INT_INTERFACE_R_X9Y46/INT_INTERFACE_LOGIC_OUTS_B9");
Node n = w.getNode();
assert n != null;  // Fails
```

clavin-xlnx, 2021-04-26:
This is correct.  Although, I don't believe there is a static method called `Device.getWire()`, you have to load a device first.   However, assuming you are using the same device as in #161, there is no node associated with the wire `INT_INTERFACE_R_X9Y46/INT_INTERFACE_LOGIC_OUTS_B9` in `xc7s25csga324`.  You can check with:

```
link_design -part xc7s25csga324
get_nodes -of [get_wires INT_INTERFACE_R_X9Y46/INT_INTERFACE_LOGIC_OUTS_B9]
```
which yields the following:
```
WARNING: [Vivado 12-2683] No nodes matched 'get_nodes INT_INTERFACE_R_X9Y46/INT_INTERFACE_LOGIC_OUTS_B9'
```

trharoldsen, 2021-04-26:
Thanks Chris,

Could you help me understand under what conditions I could get a wire that has no associated node.  I'm seeing this case pop up when I call `getBackwardPIPs` on a wire.  The returned PIP has a wire, but the wire has no node.  I'm not sure what exactly this means though.

```java
for (Wire w : n.getAllWiresInNode()) {
    for (PIP p : w.getBackwardPIPs()) {
        System.out.println(p.getStartWire().getNode());
        pips.add(p);
    }
}
```

Thanks

clavin-xlnx, 2021-04-27:
I'm not aware of the reason why some wires do not belong to a node.  So, I can't really predict which wires will not participate in a node.  It is possible there is a pattern, but I haven't had a chance to look deeply enough.

trharoldsen, 2021-04-28:
FYI, we tracked down this particular instance.  There is a PIP without that isn't driven by anything.  I presume as their isn't actually anything driving the PIP their isn't a node, but the PIP requires a source wire for consistency.

Closing

closed 2021-04-28 by trharoldsen