Xilinx/RapidWright #170: "Device mismatch" error when writing FPGA interchange for some Kintex 7 devices

mkurc-ant, 2021-04-30:
I've been trying to generate device resources database for a few Kintex 7 devices and I stumbled on the following exception being thrown for at least two of them:

For `xc7k325tffg900-1`:
```
Exception in thread "main" java.lang.RuntimeException: ERROR: Device mismatch: gold=6875, test=6852
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.expect(DeviceResourcesVerifier.java:90)
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:364)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
Command exited with non-zero status 1
```

and for `xc7k410tffg900-1`:
```
Exception in thread "main" java.lang.RuntimeException: ERROR: Device mismatch: gold=15701, test=15678
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.expect(DeviceResourcesVerifier.java:90)
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:364)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
Command exited with non-zero status 1
```

RapidWright was invoked with the following command:
```
$RAPIDWRIGHT_PATH/scripts/invoke_rapidwright.sh com.xilinx.rapidwright.interchange.DeviceResourcesExample ${DEVICE}
```

This happened for RapidWright SHA `1d4b105` on Ubuntu 18.04 Linux (64-bit). Java version:
```
openjdk version "1.8.0_275"
OpenJDK Runtime Environment (build 1.8.0_275-8u275-b01-0ubuntu1~18.04-b01)
OpenJDK 64-Bit Server VM (build 25.275-b01, mixed mode)
```

clavin-xlnx, 2021-05-01:
Thanks for finding this.  There was an issue with alternate site type sets being written to the wrong type in `DeviceResourcesWriter`.  In Series 7, a site of `OLOGICE2` has only one alternate (`OSERDESE2`), but `OLOGICE3` has two (`OLOGICE2` and `OSERDESE2`).  The code was not properly handling this scenario and was overwriting `OLOGICE2` with `OLOGICE3`'s alternates.

I have [committed a fix ](https://github.com/Xilinx/RapidWright/commit/38ab54ce7b0e843c9e00e467befa69b2a826c42f)to resolve this issue.

mkurc-ant, 2021-05-06:
Great. I've confirmed that the issue is no longer present and I can generate the database for the mentioned devices.

closed 2021-05-06 by mkurc-ant