Xilinx/RapidWright #171: Cap'n'proto "read limit exceeded"

mkurc-ant, 2021-04-30:
When running RapidWright to generate FPGA interchange device resource database for any fabric equal or larger than of `xczu6eg` the following exception is thrown:
```
Exception in thread "main" org.capnproto.DecodeException: Read limit exceeded.
	at org.capnproto.ReaderArena.checkReadLimit(ReaderArena.java:50)
	at org.capnproto.WireHelpers.readListPointer(WireHelpers.java:1168)
	at org.capnproto.ListFactory.fromPointerReaderRefDefault(ListFactory.java:38)
	at org.capnproto.ListFactory.fromPointerReaderRefDefault(ListFactory.java:24)
	at org.capnproto.StructReader._getPointerField(StructReader.java:173)
	at com.xilinx.rapidwright.interchange.DeviceResources$Device$SiteType$Reader.getPins(DeviceResources.java:988)
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:377)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
Command exited with non-zero status 1
```

I suspect that this is not a problem with RapidWright itself but rather a limitation of the Cap'n'proto library.

gatecat, 2021-04-30:
We should probably consider routing graph folding sooner rather than later. Even the most basic of implementations should be able to get a 5-10x reduction in data size.

mithro, 2021-04-30:
I thought that litghost had submitted a patch upstream which removed this
limit?

clavin-xlnx, 2021-04-30:
Although the device graph folding issue will prevent the largest devices from being created (on machines without infinite memory), this is bumping up against one of the [artificial limits that Cap'n Proto employs](https://capnproto.org/cxx.html#security-tips) for security reasons.  From the docs:

> Cap’n Proto automatically applies two artificial limits on messages for security reasons: a limit on nesting dept, and a limit on total bytes traversed.
> 
> * The nesting depth limit is designed to prevent stack overflow when handling a deeply-nested recursive type, and defaults to 64. If your types aren’t recursive, it is highly unlikely that you would ever hit this limit, and even if they are recursive, it’s still unlikely.
> 
> * The traversal limit is designed to defend against maliciously-crafted messages which use pointer cycles or overlapping objects to make a message appear much larger than it looks off the wire. While cycles and overlapping objects are illegal, they are hard to detect reliably. Instead, Cap’n Proto places a limit on how many bytes worth of objects you can dereference before it throws an exception. This limit is assessed every time you follow a pointer. By default, the limit is 64MiB (this may change in the future). StreamFdMessageReader will actually reject upfront any message which is larger than the traversal limit, even before you start reading it.
> 
> * If you need to write your code in such a way that you might frequently re-read the same pointers, instead of increasing the traversal limit to the point where it is no longer useful, consider simply copying the message into a new MallocMessageBuilder before starting. Then, the traversal limit will be enforced only during the copy. There is no traversal limit on objects once they live in a MessageBuilder, even if you use .asReader() to convert a particular object’s builder to the corresponding reader type.

Short answer, is to just bump up the number on this [line](https://github.com/Xilinx/RapidWright/blob/e81baef1225e50107f6ede6b744f5915fa2ea76d/src/com/xilinx/rapidwright/interchange/DeviceResourcesVerifier.java#L122).  However, there may be ways to decrease the total byte read based on the way the code is reading the file.  I have tried to minimize this number, but I may have missed some optimizations that might still be present.  I'll find a suitable number for now and check it in, however, when the graph folding code is ready, the number might be able to be reduced.

clavin-xlnx, 2021-04-30:
I've [increased the traversal limit](https://github.com/Xilinx/RapidWright/commit/f73818ce4c4dc1b5265e94cdac74a31680e21e61) and it works for xczu6eg.

mkurc-ant, 2021-05-06:
@clavin-xlnx Thanks. This one no longer shows up. But there's another one for even bigger devices that I've created an issue for: #176

closed 2021-05-06 by mkurc-ant