Xilinx/RapidWright #176: IOException thrown from org.capnproto.Serialize.read()

mkurc-ant, 2021-05-06:
This is a followup (presumably) issue of https://github.com/Xilinx/RapidWright/issues/171.

With #171 fixed when trying to generate device resources database for some of the largest FPGAs the following exception is now thrown:
```
Exception in thread "main" java.io.IOException: too many segments
	at org.capnproto.Serialize.read(Serialize.java:68)
	at com.xilinx.rapidwright.interchange.Interchange.readInterchangeFile(Interchange.java:76)
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:124)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
```

I've confirmed the issue for the following devices:
 - `xczu11eg-ffvc1156-1-e`
 - `xczu15eg-ffvc900-1-e`
 - `xczu17eg-ffvb1517-1-e`
 - `xczu19eg-ffvb1517-1-e`
 - `xc7vx550tffg1927-1`
 - `xc7vx690tffg1927-1`
 - `xc7vx980tffg1930-1`

clavin-xlnx, 2021-05-12:
The bug was in capnproto-java and the author kindly fixed the issue and created an updated release which can be downloaded here:
```
wget https://repo1.maven.org/maven2/org/capnproto/runtime/0.1.6/runtime-0.1.6.jar -O $(RAPIDWRIGHT_PATH)/jars/runtime-0.1.6.jar
```
You'll have to update the `CLASSPATH` environment variable to point to the new 0,1.6 jar instead of 0.1.4.  Re-running with the update appears to solve the `too many segments` issue.

clavin-xlnx, 2021-05-12:
Correction, the issue is solved for at least `xczu11eg-ffvc1156-1-e`, but I get another error with `xczu19eg-ffvb1517-1-e`:
```
Exception in thread "main" java.lang.IllegalArgumentException: capacity < 0: (-1862524376 < 0)
	at java.base/java.nio.Buffer.createCapacityException(Buffer.java:256)
	at java.base/java.nio.ByteBuffer.allocate(ByteBuffer.java:347)
	at org.capnproto.Serialize.makeByteBuffer(Serialize.java:34)
	at org.capnproto.Serialize.read(Serialize.java:89)
	at com.xilinx.rapidwright.interchange.Interchange.readInterchangeFile(Interchange.java:76)
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:124)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
```

mithro, 2021-05-12:
That looks like another integer overflow somewhere?

clavin-xlnx, 2021-05-12:
@mithro - Indeed, I've tracked it down to the usage of a Java [`ByteBuffer`](https://docs.oracle.com/en/java/javase/16/docs/api/java.base/java/nio/ByteBuffer.html), which only takes an integer (Java does not support unsigned) to allocate the in-memory buffer, thus it is limited to 2^31-1 bytes.  

In the case of the `xczu19eg`, the total number of bytes in the message has exceeded 2^31-1 bytes per [this line of Cap'n Proto Java](https://github.com/capnproto/capnproto-java/blob/a6c52407909191d8aabab83f158a5dbcb286ff45/runtime/src/main/java/org/capnproto/Serialize.java#L89):
```
2432463568 == (totalWords=304057946) * (Constants.BYTES_PER_WORD=8)
```
In order to support these larger devices with our existing schema, it would seem like Cap'n Proto Java will need to use multiple `ByteBuffer` objects in this instance.

mithro, 2021-05-12:
Looks like another bug for upstream Cap'n Proto?

clavin-xlnx, 2021-05-13:
Another integer overflow issue has been found for the `xc7v2000t`:
```
Exception in thread "main" java.lang.IndexOutOfBoundsException
	at java.base/java.nio.Buffer.checkIndex(Buffer.java:688)
	at java.base/java.nio.HeapByteBuffer.putInt(HeapByteBuffer.java:408)
	at org.capnproto.WirePointer.setKindAndTarget(WirePointer.java:49)
	at org.capnproto.WireHelpers.allocate(WireHelpers.java:85)
	at org.capnproto.WireHelpers.initStructListPointer(WireHelpers.java:506)
	at org.capnproto.StructList$Factory.initFromPointerBuilder(StructList.java:81)
	at org.capnproto.StructList$Factory.initFromPointerBuilder(StructList.java:1)
	at org.capnproto.StructBuilder._initPointerField(StructBuilder.java:194)
	at com.xilinx.rapidwright.interchange.DeviceResources$Device$Builder.initWires(DeviceResources.java:718)
	at com.xilinx.rapidwright.interchange.DeviceResourcesWriter.writeAllWiresAndNodesToBuilder(DeviceResourcesWriter.java:755)
	at com.xilinx.rapidwright.interchange.DeviceResourcesWriter.writeDeviceResourcesFile(DeviceResourcesWriter.java:269)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:29)

```

clavin-xlnx, 2021-05-13:
@mithro @mkurc-ant @gatecat 
It seems like the [latest issue](https://github.com/capnproto/capnproto-java/issues/100) is starting to bump up against the fundamental limitations of Cap'n Proto and we may need to consider changing the schema to accommodate.  The issue is that in a xc7v2000t, the [current schema](https://github.com/SymbiFlow/fpga-interchange-schema/blob/d9c7a0b366f8ad73d867211bd841fc11efd3b124/interchange/DeviceResources.capnp#L89) keeps a `Wire` array of all the wires in the device which is 346,319,666 for the 2000t.  The issue is that it takes 8 bytes per `Wire` entry yielding a number in between 2^31-1 and 2^32-1, which for Java is more difficult as it doesn't support unsigned integers.  Ultimately, Cap'n Proto won't let you have contiguous segments of more than 2^32-1, so, I think we will run into this issue again with the even larger devices Xilinx has that I have yet to try.  

The folded routing approach will hopefully resolve this limitation, but might still be a while before it gets implemented.  Thoughts?

gatecat, 2021-05-13:
A suggestion would be to remove the global `wires` array altogether; make the `wires` array inside `Node` a list of `(tile, wire)` tuples instead; and move the wire type data from `Wire` into the tile type data. I don't know how this would affect overall size and whether it would hit other issues, I can give it a try tomorrow though.

gatecat, 2021-05-13:
Anything more sophisticated than that, such as deduplicating common Node 'shapes', could get some significant improvements fairly quickly without the full complexity of folding, but are probably not worth it for an interim fix pre-full-folding unless the capnp limits do prove to be a big issue.

mithro, 2021-05-13:
We **do** need to support "flattened" structure were every node in unique.

mkurc-ant, 2021-05-14:
The simplest solution I can think of it to just split the list of nodes/wires into multiple ones. Or make it a list of lists. Then it would be up to a tool that reads/writes interchange to join them together. Not a very elegant solution though

gatecat, 2021-05-14:
I'm currently playing with a simple deduplication of node shapes based on relative coordinates:

https://github.com/gatecat/RapidWright/tree/dedup-poc
https://github.com/gatecat/fpga-interchange-schema/tree/dedup-poc

based on some previous work for nextpnr-xilinx that I never finished.

This doesn't force anyone to use deduplication, as they can have one node shape per node if they want and all the extra work they've got is one extra layer of indirection and having to use relative instead of absolute tile coordinates. It also is going to be significantly inferior to proper folding, but nonetheless hopefully a decent boost to more complex devices where most of the nodes follow the same pattern. I'm currently trying to collect some results on big devices. 

Is this something we'd be interested in pursuing as an interim measure to full folding?

gatecat, 2021-05-14:
Initial results with my deduplication proof-of-concept for the xcvu19p-fsva3824-2-e, which I believe is as big as it gets (and well beyond what nextpnr can currently target in any case :smiley_cat:) :

```
==============================================================================
==               Device Resources Dump: xcvu19p-fsva3824-2-e                ==
==============================================================================
             Load Device:     2.182s    560.313MBs
           populateEnums:     1.269s    129.294MBs
               SiteTypes:     0.122s      2.873MBs
               TileTypes:     0.521s      8.374MBs
                   Tiles:     0.770s    112.005MBs
    discovering nodes...
INFO: Building uncommon Wire->Node cache...
      This might take a few seconds for large devices on the first call.  
      It is generally triggered when getting the Node from an uncommon Wire object.  
      To avoid printing this message, set Device.QUIET_MESSAGE=true or set the ENVIRONMENT variable RW_QUIET_MESSAGE=1.
INFO: Finished building uncommon Wire->Node cache
    writing 224351820 nodes...
    writing 176949 node shapes...
             Wires&Nodes:   277.896s   4344.228MBs
            Prims&Macros:     0.359s     61.242MBs
    Cell <-> BEL pin map:   684.259s      1.186MBs
                Packages:     0.012s      0.289MBs
```

(it failed at the point of constant export).

Along the way I believe I found some unrelated implementation issues in the current interchange export that make node discovery much slower than needed, when comparing it to my old nextpnr-xilinx code. Unfortunately my first attempt at a fix seems to have broken constant export, so I need to work out what's going on there to get a full result.

gatecat, 2021-05-14:
Full results for the `xc7v2000t` which doesn't seem to hit the constant issue (and which I'm starting to think could actually be unrelated to that patch but a vu19p-specific issue).

```
==============================================================================
==                Device Resources Dump: xc7v2000tflg1925-1                 ==
==============================================================================
             Load Device:     1.313s    337.497MBs
           populateEnums:     0.622s     64.303MBs
               SiteTypes:     0.090s      0.975MBs
               TileTypes:     0.210s     10.063MBs
                   Tiles:     0.291s     20.005MBs
    discovering nodes...
INFO: Building uncommon Wire->Node cache...
      This might take a few seconds for large devices on the first call.  
      It is generally triggered when getting the Node from an uncommon Wire object.  
      To avoid printing this message, set Device.QUIET_MESSAGE=true or set the ENVIRONMENT variable RW_QUIET_MESSAGE=1.
INFO: Finished building uncommon Wire->Node cache
    writing 57693368 nodes...
    writing 77284 node shapes...
             Wires&Nodes:    73.342s   2084.649MBs
            Prims&Macros:     0.346s     76.565MBs
    Cell <-> BEL pin map:   207.096s      0.982MBs
                Packages:     0.012s      0.089MBs
               Constants:    26.156s      0.012MBs
              Wire Types:     0.001s      0.009MBs
                 Strings:     0.198s      0.002MBs
              Write File:    34.264s      0.003MBs
```

gatecat, 2021-05-14:
Yeah, wires are the connections within tiles (i.e. the things that connect to pips and bel pins) and nodes directly connect wires togther across tiles.

nelsobe, 2021-05-18:
We have a couple of de-duplicated node/wire structures our students have been working with (started with some ideas from litghost before his departure and has continued).  They are wrapped in capnproto for serialization and so should be easy to compare to @gatecat's results from above. The original goal was a replacement of VPR's rrgraph's nodes/wires structure with a folded structure.  

I have added a student (@ryancj14) who is now starting to work on pulling the needed information from Interchange (as opposed to how litghost originally did it which did not use Interchange).  It makes sense to compare what we get to @gatecat's results above (and anything else that has been tried).  We could then see if or how the various approaches could be combined to lead to a solution to this problem.  And, solving this issue will also help with our original charge to provide a reduced rrgraph.   

Since we plan to take this on as a task we would love to visit with those interested in the problem to understand (a) what has been done so (b) we can contribute and (c) not duplicate work or (d) push in directions that aren't helpful.  Anyone interested in such a quick visit?  I feel like we could strongly benefit from such a discussion before we get too far into it.

clavin-xlnx, 2022-02-04:
Fundamentally, operating with a single message to capture the entire device exceeds the limits of Cap'n Proto (Java).  At this point we have two options, reduce message size (following @nelsobe suggestion and @gatecat prototype PR https://github.com/chipsalliance/fpga-interchange-schema/pull/54) or we modify the Interchange to support multiple messages. Closing for now.

closed 2022-02-04 by clavin-xlnx