Xilinx/RapidWright #178: Versal interchange export

gatecat, 2021-05-08:
Attempting to export an interchange file for the device `xcvm1902-vsvd1760-2MP-e-S`, I hit the following issues - this isn't critical, but given the new carry structure in the Versal devices it'd probably be good to have them working at some point to ensure whatever we design is capable of supporting them.

1. The `DSP_FP_SRCMX_OPTINV` bel has `canInvert()` true but `getInvertingPin()` returning `null`, because it's not a conventional inverter but designed to have an actual cell placed on it. Probably the fix for it is not writing this bel as an inverter, because it will always have a cell placed on it in reality.
2. `PseudoPIPHelper` `getUsedBELPins()` is returning null for some route-through PIPs associated with the AIE, like `AIE_NOC_FROM_PL_2_PIN` to `AIE_NOC_TO_AIE_2_PIN` in the `AIE_INTF_B2_CORE` tile type.
3. Switching to the `xcvm1802-vsvd1760-2MP-e-S` device, which doesn't have those tiles, wire/node export fails with an internal null pointer exception in `wire.getNode()` for `INTF_GT_TR_TILE_X112Y330/INTF_MUX2_TMR_GREEN_TMR_DFT`.

clavin-xlnx, 2021-05-18:
Thanks @gatecat for pushing on this.  In the background I believe I have fixed the 3 issues you have raised and fixed a few more.  I'm working on a release (2020.2.5) which will contain the fixes, I can get the device model to build but it fails in verification:
```
               Constants:   383.537s      0.012MBs
              Wire Types:     0.008s      0.007MBs
                 Strings:     0.249s      0.001MBs
              Write File:    88.675s      0.003MBs
INFO: Building uncommon Wire->Node cache...
      This might take a few seconds for large devices on the first call.  
      It is generally triggered when getting the Node from an uncommon Wire object.  
      To avoid printing this message, set Device.QUIET_MESSAGE=true or set the ENVIRONMENT variable RW_QUIET_MESSAGE=1.
INFO: Finished building uncommon Wire->Node cache
Exception in thread "main" Listening for transport dt_socket at address: 8001
java.lang.RuntimeException: Tile BLI_LS_CORE_X90Y335 node HSR_GRP0_A_BLI_LOGIC_OUTS_CTRL0 tie mismatch!
	at com.xilinx.rapidwright.interchange.ConstantDefinitions.verifyConstants(ConstantDefinitions.java:233)
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:540)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
	at com.xilinx.rapidwright.examples.Runner.main(Runner.java:23)
```
The issue is that Versal has a few things that has broken some assumptions in the code.  I think the Interchange schema is probably fine for now, but I still have some work to do to update things.  At least two things have stood out:

1. Versal allows site pips to share input pins (two site pips can originate from the same site pip).
2. Some wires in a tile type can be tied in some contexts but not tied in others.  Some of the verification code was built to rely on the assumption that any wire in a tile type will always be tied regardless of context.  It does allow for exceptions (in some contexts VCC, others GND), we just need to broaden that case to sometimes its also untied.

clavin-xlnx, 2021-05-25:
@gatecat I just pushed a [fix (e1c9742)](https://github.com/Xilinx/RapidWright/commit/e1c9742ba9549ac6ccb22c260cf1b86cd751d513) that should hopefully allow you to fully build the two mentioned Versal devices.  

Note: for future reference, I believe the part `xcvm1902-vsvd1760-2MP-e-S` was actually intended to be `xcvc1902-vsvd1760-2MP-e-S` .

clavin-xlnx, 2021-06-23:
The 2020.2.6 release contains this fix and should resolve the 3 cited issues.

closed 2021-06-23 by clavin-xlnx