Xilinx/RapidWright #181: Node tied value mismatch on VU19P

gatecat, 2021-05-14:
While doing some large device testing for #176 and investigating the failure in the constant export, it seems like there is a discrepancy in the RapidWright node tied value data.

Consider the following MCVE:

```java
package net.gatecat.fpga;

import java.io.File;
import java.io.IOException;

import com.xilinx.rapidwright.device.Device;
import com.xilinx.rapidwright.device.Node;

public class VU19PNode {
	public static void main(String[] args) throws IOException {
		Device d = Device.getDevice("xcvu19p-fsva3824-2-e");
		Node n = d.getNode("INT_IBRK_FSR2IO_X0Y1040/IO_TO_CTR_FT0_3");
		System.out.println("GND: " + n.isTiedToGnd());
	}
}
```

it prints `GND: false`.

Whereas Vivado shows the same node as being tied to ground both in the UI and in Tcl:
[image]

```tcl
get_property IS_GND [get_nodes INT_IBRK_FSR2IO_X0Y1040/IO_TO_CTR_FT0_3]
1
```

clavin-xlnx, 2021-05-19:
The [2020.2.5](https://github.com/Xilinx/RapidWright/releases/tag/v2020.2.5-beta) release should fix this issue.

gatecat, 2021-05-19:
Thanks, this is working now.

closed 2021-05-19 by gatecat