Xilinx/RapidWright #183: Missing pin mappings for CARRY4

gatecat, 2021-05-18:
Not sure if I'm doing something wrong, as I'm surprised this hasn't been hit by someone else yet?

Consider:

```java
package net.gatecat.fpga;

import java.io.File;
import java.io.IOException;
import java.util.stream.Collectors;
import java.util.Map.Entry;
import java.util.Map;

import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Unisim;

public class PinMap {
	public static void main(String[] args) throws IOException {
		Design des = new Design("top", "xc7a35tcsg324-1");
		Cell c = des.createAndPlaceCell("carry_i", Unisim.CARRY4, "SLICE_X0Y0/CARRY4");
		for (Entry<String,String> m : 
			c.getPinMappingsP2L().entrySet()
			.stream().sorted(Map.Entry.<String, String>comparingByKey())
			.collect(Collectors.toList())) {
			System.out.printf("%s %s\n", m.getKey(), m.getValue());
		}
	}
}
```

All this prints is
```
CIN CI
```

No mappings are included for the `S`, `DI`, `O`, or `CO` buses. The same problem seems to affect the interchange output, at least at a glance.

acomodi, 2021-05-18:
@gatecat Might this be due to the fact that the mapping is applied only when required?

For instance, all the `S`, `DI`, `O` and `CO` buses are shared between cell and BEL definition (at least it seems to be so), while `CI` -> `CIN` needs an explicit mapping.

IMO, even if the above is true, it might be better to always have an explicit mapping for each port.

gatecat, 2021-05-18:
Yes, I would prefer an explicit mapping too - or at the very least some clear documentation on when and how the mapping can be elided.

kowalewskijan, 2021-05-18:
I stumbled upon it yesterday and I assumed like @acomodi said that only `CI->CIN`  needs mapping. However, currently I stuck with a problem regarding finding a wire between a CARRY4 and another cell. In the Nexpnr it is assumed that if there is no pin in the CellBelMapping there is no corresponding wire. This info should be provided during chipdb generation if elided in RW, otherwise RW should provide it.

gatecat, 2021-05-18:
Other cells don't seem to elide pins with the same name. A more generic version of the above code is:

```java
package net.gatecat.fpga;

import java.io.File;
import java.io.IOException;
import java.util.stream.Collectors;
import java.util.Map.Entry;
import java.util.Map;

import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Unisim;

public class PinMap {
	public static void main(String[] args) throws IOException {
		Design des = new Design("top", args[0]);
		Cell c = des.createAndPlaceCell("carry_i", Unisim.valueOf(args[1]), args[2]);
		for (Entry<String,String> m : 
			c.getPinMappingsP2L().entrySet()
			.stream().sorted(Map.Entry.<String, String>comparingByKey())
			.collect(Collectors.toList())) {
			System.out.printf("%s %s\n", m.getKey(), m.getValue());
		}
	}
}
```

run with `xc7a35tcsg324-1 FDRE SLICE_X0Y0/AFF` as arguments gives:

```
CE CE
CK C
D D
Q Q
SR R
```

`CE`, `D`, and `Q` are all named the same but included explicitly in the mapping.

clavin-xlnx, 2021-05-19:
I'm not sure why those pin mappings are missing.  Even when I create a new `CARRY4` in Vivado in an empty context, it still has several more pin mappings.  Although, strangely, it doesn't have a mapping for `CI`.
```
link_design -part xc7a35tcsg324-1
create_cell -reference CARRY4 carry4Inst
place_cell carry4Inst SLICE_X0Y0/CARRY4
```
[image]
When I look at the `CARRY8` for UltraScale devices, there seems to be the correct number of mappings.  Perhaps this could just be a place where I need to supplement RapidWright as I don't seem to be getting the proper information.  Should I add the pin mappings for the Series 7 `CARRY4` based on what Vivado produces?  Or, are there other pin mappings that we should be expecting as well?

gatecat, 2021-05-19:
> Although, strangely, it doesn't have a mapping for CI

This might be due to the choice of both CI (for carries coming from the chain) and CYINIT (for carry in from fabric or constant) - aiui only one of the two cell pins, and hence bel pins - can only be used at once. But including both in the mapping should be fine too.

> Or, are there other pin mappings that we should be expecting as well?

I don't think we've seen any other cases of this.

clavin-xlnx, 2021-05-19:
I've made a patch to include the `CARRY4` mappings, you can update your RapidWright with the following commands:
```
cd $RAPIDWRIGHT_PATH
wget https://github.com/Xilinx/RapidWright/releases/download/v2020.2.5-beta/rapidwright_api_lib-2020.2.5-patch1.zip
unzip -o rapidwright_api_lib-2020.2.5-patch1.zip
```
If we are satisfied with this mapping set, I can add it permanently to the next release.

kowalewskijan, 2021-05-20:
LGMT - the patch works for me

clavin-xlnx, 2021-06-23:
This fix is included in [2020.2.6](https://github.com/Xilinx/RapidWright/releases/tag/v2020.2.6-beta).

closed 2021-06-23 by clavin-xlnx