Xilinx/RapidWright #186: Missing connections

trharoldsen, 2021-05-20:
The following code reports that the wire `BRAM_INT_INTERFACE_L_X6Y31/INT_INTERFACE_LOGIC_OUTS_L13` has no connections.  Unless I'm misunderstanding how connections should work, it should have 1 or more connections going out to the switchbox.

Version: RapidWright 2020.2.5
Part: xc7s25csga324

```
Device device = Device.getDevice("xc7s25csga324-1");
Wire wire = device.getWire("BRAM_INT_INTERFACE_L_X6Y31/INT_INTERFACE_LOGIC_OUTS_L13");
System.out.println(wire.getTile().getWireConnections(wire.getWireIndex()));
```

clavin-xlnx, 2021-05-20:
Thanks for pointing this out.  I have created a patch that should resolve the issue.  If it satisfies the issue, I can add it to the next release.  To update your RapidWright with the patch run the following:
```
cd $RAPIDWRIGHT_PATH
wget https://github.com/Xilinx/RapidWright/releases/download/v2020.2.5-beta/rapidwright_api_lib-2020.2.5-patch2.zip
unzip -o rapidwright_api_lib-2020.2.5-patch2.zip
```

clavin-xlnx, 2021-06-23:
This fix is included in [2020.2.6](https://github.com/Xilinx/RapidWright/releases/tag/v2020.2.6-beta).

closed 2021-06-23 by clavin-xlnx