Xilinx/RapidWright #188: EDIFNetlist.generateParentNetMap() StringIndexOutOfBoundsException when '/' used in cells/nets

DallonTG, 2021-06-08:
On the latest version of RapidWright, I encounter this exception whenever EDIFNetlist.generateParentNetMap() is used on a flat design that has the hiearchy separator ('/') in cell and/or net names. Obviously it is non-ideal to have slashes in cell/net names. For my use-case, this occurs because Vivado infers a DSP in my design that I synthesize with "-flatten_hierarchy full" and uses a '/' in the inferred cell name and nets. I think this bug was introduced some time after the 2019.1.2 release.

```
Exception in thread "main" java.lang.StringIndexOutOfBoundsException: begin 0, end -1, length 19
	at java.base/java.lang.String.checkBoundsBeginEnd(String.java:3720)
	at java.base/java.lang.String.substring(String.java:1909)
	at com.xilinx.rapidwright.edif.EDIFNetlist.getNetAliases(EDIFNetlist.java:994)
	at com.xilinx.rapidwright.edif.EDIFNetlist.generateParentNetMap(EDIFNetlist.java:1159)
	at com.xilinx.rapidwright.edif.EDIFNetlist.getParentNetMap(EDIFNetlist.java:1101)
```

clavin-xlnx, 2021-06-14:
Thanks for pointing out this error.  Would you be able to share a design (any design) that exhibits this issue?

DallonTG, 2021-07-05:
I thought I had isolated the issue to what I described, but I have unfortunately been able to replicate the issue. The problem may have been caused by me renaming some nets, etc. using TCL in Vivado prior to importing the design into RapidWright. I will reopen this issue if I encounter it again (and can provide an example).

PR #207 (2021-07-27): Use EDIFCellInst instead of String hierarchy in EDIFHier classes
Previously, `EDIFHierCellInst` stored all hierarchy as a single String. When we accessed hierarchy info, we had to re-parse the string every time. This PR replaces the internal representation of `EDIFHierCellInst` with an Array of `EDIFCellInst`.

This provides a speedup for `EDIFNetlist.getParentNetMap` while simultaneously decreasing memory usage. It also fixes #188.

This PR is dependent on #198. Github's view for this PR will include all of #198's diff until that one is merged. As #198 currently fails testing, this one does a [... cut]
closed 2021-07-05 by DallonTG