Xilinx/RapidWright #190: java.lang.IllegalAccessError due to internal code access by Kryo with Java 16

gatecat, 2021-06-18:
Running anything with RapidWright on Java 16 fails with an `IllegalAccessError`:

```
% ./scripts/invoke_rapidwright.sh com.xilinx.rapidwright.examples.Lesson1
Exception in thread "main" java.lang.IllegalAccessError: class com.esotericsoftware.kryo.unsafe.UnsafeUtil (in unnamed module @0x2baf3d81) cannot access class sun.nio.ch.DirectBuffer (in module java.base) because module java.base does not export sun.nio.ch to unnamed module @0x2baf3d81
	at com.esotericsoftware.kryo.unsafe.UnsafeUtil.<clinit>(UnsafeUtil.java:110)
	at com.esotericsoftware.kryo.unsafe.UnsafeInput.readInt(UnsafeInput.java:92)
	at com.xilinx.rapidwright.util.FileTools.readStringArray(FileTools.java:279)
	at com.xilinx.rapidwright.device.PartNameTools.<clinit>(PartNameTools.java:51)
	at com.xilinx.rapidwright.design.Design.setPartName(Unknown Source)
	at com.xilinx.rapidwright.design.Design.<init>(Unknown Source)
	at com.xilinx.rapidwright.examples.Lesson1.main(Lesson1.java:49)
```

This isn't blocking me as it's easy enough to downgrade Java but it feels like something that should be fixed long term. It's slightly unclear if there's currently a good alternative other than moving away from Kryo's unsafe buffers which would likely have its own downsides...

gatecat, 2021-06-18:
A workaround is to add the `--add-exports` command line argument to Java:

```
./scripts/invoke_rapidwright.sh --add-exports=java.base/sun.nio.ch=ALL-UNNAMED com.xilinx.rapidwright.examples.Lesson1
```

Ideally `./scripts/invoke_rapidwright.sh` would add this automatically (at least until another solution is found, perhaps), but I think some older versions of Java might not have it so it may need to be conditional on Java version...

Xiretza, 2021-08-01:
This also affects the unit tests, except the `--add-exports` option can't simply be passed to gradle. I've disabled the unit tests in the [AUR package](https://aur.archlinux.org/packages/rapidwright/) for now (using `--exclude-task test`), but I would love to hear a better alternative if someone is more familiar with gradle.

clavin-xlnx, 2021-11-03:
This should be resolved in 2021.1.2, by default, it will now switch away from using the Unsafe I/O classes in Kryo when using Java 16 and above.

closed 2021-11-03 by clavin-xlnx