Xilinx/RapidWright #192: Extra Pips Reported in Interchange Database?

reillymck, 2021-06-18:
It appears that interchange reports pips that do not appear in Vivado for the xc7a100tcsg324-1.
One example of this occurs in the LIOI3_X0Y197 tile. Interchange seems to document these pips:
```
Pip wire0: IOI_IMUX24_0 wire1: IOI_IDELAYCTRL_RST
Pip wire0: IOI_IDELAYCTRL_OUTN65 wire1: IOI_LOGIC_OUTS16_1
Pip wire0: IOI_IDELAYCTRL_DNPULSEOUT wire1: IOI_LOGIC_OUTS13_0
Pip wire0: IOI_IDELAYCTRL_OUTN1 wire1: IOI_LOGIC_OUTS13_1
Pip wire0: IOI_IDELAYCTRL_UPPULSEOUT wire1: IOI_LOGIC_OUTS16_0
Pip wire0: IOI_IDELAYCTRL_RDY wire1: IOI_LOGIC_OUTS22_1
```
However, when querying Vivado with a command like:
```
select_objects [get_pips LIOI3_X0Y197/LIOI3.IOB_DIFFO_OUT0->IOB_DIFFO_IN1]
```
the following error is produced:
```
WARNING: [Common 17-346] No pips matched 'LIOI3_X0Y197/LIOI3.IOB_DIFFO_OUT0->IOB_DIFFO_IN1'
```
Is there something we are missing here? This seems to occur a little over 900 times for this part.

Also, this may not be important but the ISE XDLRC file does not have record of these pips either.

clavin-xlnx, 2021-06-21:
For the first 6 example PIPs you list, I am able to query Vivado (2020.2) and find them:

```
% link_design -part xc7a100tcsg324-1
% get_pips {LIOI3_X0Y197/LIOI3.IOI_IMUX24_0->IOI_IDELAYCTRL_RST LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_OUTN65->IOI_LOGIC_OUTS16_1 LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_DNPULSEOUT->IOI_LOGIC_OUTS13_0 LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_OUTN1->IOI_LOGIC_OUTS13_1 LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_UPPULSEOUT->IOI_LOGIC_OUTS16_0 LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_RDY->IOI_LOGIC_OUTS22_1}
LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_DNPULSEOUT->IOI_LOGIC_OUTS13_0 LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_OUTN1->IOI_LOGIC_OUTS13_1 LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_OUTN65->IOI_LOGIC_OUTS16_1 LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_RDY->IOI_LOGIC_OUTS22_1 LIOI3_X0Y197/LIOI3.IOI_IDELAYCTRL_UPPULSEOUT->IOI_LOGIC_OUTS16_0 LIOI3_X0Y197/LIOI3.IOI_IMUX24_0->IOI_IDELAYCTRL_RST
```

I am unable to find any reference to the PIP `LIOI3_X0Y197/LIOI3.IOB_DIFFO_OUT0->IOB_DIFFO_IN1` you mentioned in RapidWright, Vivado or otherwise.  In fact, I do not see those wires in the tile nor site wires.  To investigate further, could you provide some code in RapidWright that prints out the PIP you have found to be missing?

reillymck, 2021-07-13:
Sorry about this. My mistake was that I did not notice that some pips have "->" and some pips have "->>" in Vivado, so when I tried to do a lookup, some of them failed.

closed 2021-07-13 by reillymck