Xilinx/RapidWright #193: Macro expansion map needs to be completed and parameter-dependent

gatecat, 2021-06-25:
See the discussion in SymbiFlow/fpga-interchange-schema#40 and SymbiFlow/fpga-interchange-tests#42

Currently a single entry is hard-coded in the macro expansion map: https://github.com/Xilinx/RapidWright/blob/96c39ae8380bb1f9ecf855459889bc44096e86a9/src/com/xilinx/rapidwright/edif/EDIFNetlist.java#L97

This mapping is actually only valid for pseudo-differential IO types, my understanding is that true differential IO types like LVDS expand using the `OBUFTDS` macro instead. A parameter match will be needed here for all the pseudo differential types.

Similarly this list needs expanding to other primitive types with the same behaviour like `OBUFDS` and `IOBUFDS`.

clavin-xlnx, 2021-06-25:
Happy to accommodate these changes.  I'll have to investigate how to systematically harvest the variety of scenarios that need to be encoded here.  The single entry was added simply because that was the first case we ran into, but sounds like there are several more.  If you happen to have any scripts that can enumerate these cases, feel free to share.

gatecat, 2021-06-28:
> If you happen to have any scripts that can enumerate these cases, feel free to share.

Unfortunately not, this doesn't seem to be something that Vivado exposes very well.

clavin-xlnx, 2021-07-22:
Closing for now, please re-open if any deficiencies are found.

closed 2021-07-22 by clavin-xlnx