Xilinx/RapidWright #194: xczu3eg-sbva484-1-i interchange fails verification due to ground tie issue

gatecat, 2021-06-28:
Run
```
./scripts/invoke_rapidwright.sh com.xilinx.rapidwright.interchange.DeviceResourcesExample xczu3eg-sbva484-1-i
```

fails with

```
Exception in thread "main" java.lang.RuntimeException: Tile HPIO_RIGHT_X32Y90 node OPTION2IOB_TX_DATA_OPT2 GND tie mismatch! false != true
	at com.xilinx.rapidwright.interchange.ConstantDefinitions.verifyConstants(ConstantDefinitions.java:244)
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:539)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
```

gatecat, 2021-06-28:
In case it helps, the correct tie value of the node according to Vivado is Vcc:

```
get_property IS_VCC [get_nodes HPIO_RIGHT_X32Y90/OPTION2IOB_TX_DATA_OPT2]
1
```

clavin-xlnx, 2021-06-29:
Just checked in a fix ([b106c3a](https://github.com/Xilinx/RapidWright/commit/b106c3ab421fe7b9b5986c42a234b3fba2190e3b)) for this issue, should be resolved.

gatecat, 2021-06-30:
Thanks, this is looking good now!

closed 2021-06-30 by gatecat