Xilinx/RapidWright #195: Irregularity with UltraScale+ disconnected bel pin

gatecat, 2021-06-28:
Consider the xczu7ev bel pin `BITSLICE_RX_TX_X0Y305/RXTX_BITSLICE/RX_DIV2_CLK_Q`. According to Vivado it's not connected to any site wires:
[image]

However, the following RapidWright snippet shows it connected to site wire 0 (which I suspect is just intended to be a placeholder, but 0 is a valid site wire), while not appearing in the list of bel pins when querying the pins on the site wire:

```java
package net.gatecat.fpga;

import java.io.File;
import java.io.IOException;

import com.xilinx.rapidwright.device.*;

public class BitslicePin {
	public static void main(String[] args) throws IOException {
		Device d = Device.getDevice("xczu7ev-ffvc1156-2-e");
		Site s = d.getSite("BITSLICE_RX_TX_X0Y305");
		BEL b = s.getBEL("RXTX_BITSLICE");
		BELPin pin = b.getPin("RX_DIV2_CLK_Q");
		System.out.println("site wire index: " + pin.getSiteWireIndex());
		System.out.println("site wire name according to pin: " + pin.getSiteWireName());
		for (BELPin otherPin : s.getBELPins(pin.getSiteWireName())) {
			System.out.println("connected bel pin: " + otherPin.getBELName() + "/" + otherPin.getName());
		}
	}
}
```

prints:
```
site wire index: 0
site wire name according to pin: BS2CTL_IDELAY_DELAY_FORMAT
connected bel pin: RXTX_BITSLICE/RX_BIT_CTRL_OUT12
connected bel pin: BS2CTL_IDELAY_DELAY_FORMAT/BS2CTL_IDELAY_DELAY_FORMAT
```

note `RXTX_BITSLICE/RX_DIV2_CLK_Q` not showing up as a "connected bel pin".

gatecat, 2021-06-28:
Although there's definitely an irregularity here, it turns out the interchange is handling this correctly and the actual bug I was dealing with was https://github.com/YosysHQ/nextpnr/pull/735, so this isn't blocking me.

clavin-xlnx, 2021-11-17:
This should be resolved in 2021.2.0.

closed 2021-11-17 by clavin-xlnx