Xilinx/RapidWright #196: [Interchange] Missing handling of alternate site types

acomodi, 2021-06-30:
While testing `IOSERDESes` support, I bumped into a possibly missing handling of the alternative site types when generating the DCP from a physical netlist.

The IOLOGIC sites do in fact allow for three different site types and the IOSERDESes are not the primary types.

This is an example of incorrect DCP, starting from a physical netlist (included in the tar): [serdes.tar.gz](https://github.com/Xilinx/RapidWright/files/6740559/serdes.tar.gz)

Moreover, what was thought to be inserted within the [altSiteType](https://github.com/SymbiFlow/fpga-interchange-schema/blob/ca10e820b788925eacccbbde41ba0fb0adc510f0/interchange/PhysicalNetlist.capnp#L73) in the schema? I assume that that would be the name of the alternative site type (`I/OSERDESE2` in this case)

clavin-xlnx, 2021-06-30:
Thanks for the design example.  From what I can tell, it does not appear to be a problem with the alternate site type handling.  It appears to have more to do with the way Vivado validates the site upon load.  When I attempt to load the DCP I get the following critical warnings:

```
CRITICAL WARNING: [Constraints 18-4534] Net OFB.0__unique__1 does not exist.
CRITICAL WARNING: [Constraints 18-4534] Net SHIFTOUT1.0__unique__1 does not exist.
CRITICAL WARNING: [Constraints 18-4534] Net SHIFTOUT2.0__unique__1 does not exist.
CRITICAL WARNING: [Constraints 18-4534] Net TBYTEOUT.0 does not exist.
CRITICAL WARNING: [Constraints 18-4534] Net TFB.0 does not exist.
CRITICAL WARNING: [Constraints 18-4534] Net O.0 does not exist.
CRITICAL WARNING: [Constraints 18-4534] Net SHIFTOUT1.0 does not exist.
CRITICAL WARNING: [Constraints 18-4534] Net SHIFTOUT2.0 does not exist.
CRITICAL WARNING: [Designutils 20-2070] During XDEF restore, placement information for 2 sites failed to restore.  Please run placement and routing to complete implementation.
```

At first glance, the final warning does not seem related to the prior ones, but that seems to be the case.  Somewhere, those nets are being created and mapped to corresponding site wires, but as there is no logical net equivalent, Vivado, will abort loading the site.  I've been able to check this by removing the site routing to the non-existent nets:

```
        // Load Interchange netlist...

        SiteInst si = design.getSiteInstFromSiteName("ILOGIC_X0Y25");
        si.getNetSiteWireMap().remove("O");
        si.getNetSiteWireMap().remove("SHIFTOUT1");
        si.getNetSiteWireMap().remove("SHIFTOUT2");
        
        si = roundtrip.getSiteInstFromSiteName("OLOGIC_X0Y25");       
        si.getNetSiteWireMap().remove("OFB");
        si.getNetSiteWireMap().remove("SHIFTOUT1");
        si.getNetSiteWireMap().remove("SHIFTOUT2");
        si.getNetSiteWireMap().remove("TBYTEOUT");
        si.getNetSiteWireMap().remove("TFB");

       // Write out DCP...
```

This allows the sites ILOGIC_X0Y25/OLOGIC_X0Y25 to be populated correctly in Vivado and matches the behavior if I place them manually using Vivado's Tcl `place_cell`:
[image]

Alternatively, I can create the missing logical nets after loading the Interchange design and also get the cells to be placed:

```
        // Load Interchange netlist...

        EDIFCell top = design.getTopEDIFCell();
        for(String cellInstName : new String[] {"serdes_test.iserdes", "serdes_test.oserdes"}) {
        	Cell c = design.getCell(cellInstName);

        	if(cellInstName.equals("serdes_test.iserdes")) {
                      top.createNet("O.0").createPortInst("O", c);
                      top.createNet("SHIFTOUT1.0").createPortInst("SHIFTOUT1", c);
                      top.createNet("SHIFTOUT2.0").createPortInst("SHIFTOUT2", c);
        	} else {
                      top.createNet("OFB.0__unique__1").createPortInst("OFB", c);
                      top.createNet("SHIFTOUT1.0__unique__1").createPortInst("SHIFTOUT1", c);
                      top.createNet("SHIFTOUT2.0__unique__1").createPortInst("SHIFTOUT2", c);        
                      top.createNet("TBYTEOUT.0").createPortInst("TBYTEOUT", c);
                      top.createNet("TFB.0").createPortInst("TFB", c);
        	}
        }

       // Write out DCP...
```
[image]

So you might need to choose one of those two representations to move forward and have the design accepted correctly by Vivado.

clavin-xlnx, 2021-06-30:
Let me know if I somehow missed the issue with alternate site types, but I am assuming that the failure to load the placements/site routing correctly was the original issue.

acomodi, 2021-07-01:
@clavin-xlnx Thanks for the quick update and for investigating this. After all, this had nothing to do with alternate site types (I wonder whether the physical netlist schema really needs the altSiteType attribute).

In any case I can confirm that the problem was that some nets with no users were written in the physical netlist, causing this issue. I have opened a fix here: https://github.com/YosysHQ/nextpnr/pull/742.

closed 2021-07-01 by acomodi