Xilinx/RapidWright #199: Add missing cell pin default connections

clavin-xlnx, 2021-07-02:
Track the issue SymbiFlow/fpga-interchange-tests/#40.  The goal is to provide cell pin default connections for a new instance.

mkurc-ant, 2021-08-24:
Is there a chance for having this information included in device resource in foreseeable future? I don't think that Vivado itself currently provides that information.

clavin-xlnx, 2021-08-24:
@mkurc-ant  Please take a look at the latest commit (https://github.com/Xilinx/RapidWright/commit/c53114453da6327c9b708a8f4ccde081aa7105f4), I've attempted to populate the Interchange Device schema with cell pin static default values per @gatecat technique of extracting from Vivado.

mkurc-ant, 2021-08-25:
@clavin-xlnx Thanks, I'll look into data produced with this change.

clavin-xlnx, 2021-08-30:
Closing for now, please reopen if any issues are found with the data.

closed 2021-08-30 by clavin-xlnx