Xilinx/RapidWright #205: Case where a macro expands to a single primitive

gatecat, 2021-07-14:
The UltraScale+ has a macro `BUFG` that expands to a `BUFGCE`. However, when using this through the interchange flow, Vivado reports the expanded BUFGCE as unplaced. I think this is because it expects the BUFGCE to have the same name as the cell it replaces, rather than a hierarchical name as it would in a macro that expands to more than one cell:

```
ERROR: [DRC UNPL-1] Unplaced Cell: $auto$clkbufmap.cc:261:execute$2192 is not placed. Every cell must be placed in order to generate a valid bitstream. Please run tcl command place_design
```

The placement in the physical netlist is for the expanded cell `$auto$clkbufmap.cc:261:execute$2192/BUFGCE_INST`.

Files: 
[counter.zip](https://github.com/Xilinx/RapidWright/files/6814888/counter.zip)

If I modify nextpnr to name the expanded cell the same as the cell it expands from, as I think Vivado is expecting, RapidWright then fails to create the DCP:

```
Exception in thread "main" java.lang.RuntimeException: ERROR: Placement for macro primitive BUFG (instance $auto$clkbufmap.cc:261:execute$2192) is invalid.  Please only provide placements for the macro's children leaf cells: [].
	at com.xilinx.rapidwright.interchange.PhysNetlistReader.readPlacement(PhysNetlistReader.java:188)
	at com.xilinx.rapidwright.interchange.PhysNetlistReader.readPhysNetlist(PhysNetlistReader.java:86)
	at com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp.main(PhysicalNetlistToDcp.java:38)
```

I think there are two options here:
 - modify RapidWright to fix up the placement info before creating the DCP, so the macro rules in the interchange format are consistent even for macros that only expand to one cell
 - modify the interchange spec & nextpnr to match Vivado's behaviour here, and then modify RapidWright to accept these netlists

clavin-xlnx, 2021-07-15:
Thanks @gatecat for pointing out this discrepancy.  I think I might understand why this might be happening.  

Typically, macro expansion and collapse happens at netlist read and write time, respectively.  This is a reversible process and happens deterministically and as you have correctly pointed out, the macro leaf cells contain entries in the physical netlist as children of the macro.  For the BUFG on UltraScale+, the BUFG to BUFGCE transformation seems to be in a different category we might call "retargeted."  This appears to be a one-time process that changes or upgrades a legacy primitive to use the current architectural equivalent.  

For example, in Vivado with an UltraScale+ design open if I attempt to create a BUFG, I get the following WARNING:
```
link_design -part xczu7ev-ffvc1156-1-i
create_cell -reference BUFG myBUFG
WARNING: [Coretcl 2-1024] Primitives of type 'BUFG' are not supported by the current part and have been retargeted to 'BUFGCE'.
```
Once I write out the design with the BUFGCE, it doesn't revert back to a BUFG.  Obviously, RapidWright is not properly capturing the distinction between macros and retargeted primitives so there is a deficiency there that I will need to augment.  I can probably fix this such that the macro set is separated into two sets {macros, retargeted} so that they can be properly handled upon design load.  I imagine there are other considerations that might need to be captured since this is a different replacement type than macro expansion, we probably need a pin-remapping and possibly add properties--I'll have to investigate further all the behaviors that take place.  

Perhaps to offer the greatest compatibility and flexibility, we might consider capturing the retargeting process in the Interchange?  It doesn't seem like it would be needed immediately as RapidWright will handle the retargeting, but we might need to communicate that transformation so that it happens correctly in other tools as well.

gatecat, 2021-07-15:
Thanks for your detailed investigation!

Another option for retargets would be to write out a Verilog techmap file for Yosys that is used to permanently remap the cells after synthesis but before PnR.

We already have this manually in place for a few retarget-like transforms (actually I think these are both retargets in Vivado too): https://github.com/SymbiFlow/fpga-interchange-tests/blob/master/tests/common/remap_xc7.v

mithro, 2021-07-16:
I think retarget transforms should be the responsibility of synthesis tool?

clavin-xlnx, 2021-07-19:
I agree. Ideally, the synthesis tool would automatically perform the retargeting.  If we operate under that assumption, then the Interchange would not need to provide that transformation.

eddieh-xlnx, 2021-09-28:
@clavin-xlnx I think this particular case should be resolved by #240 and un-rigorously tested by #235?

clavin-xlnx, 2021-09-29:
Yes, I think for the BUFG case, this should be resolved.  We recently were digging deeper on the topic of macros vs. retargeted primitives.  The BUFG and CFGLUT5 both are retargeted prims if you try to instantiate them directly in Vivado.  However, if they are already instantiated in the netlist, Vivado preserves them and treats them like a macro.  #240 adds macro definitions for these two primitives so that the RapidWright behavior is the same (or attempted) as Vivado.