Xilinx/RapidWright #206: Net using concurrently pseudo PIP and LUT5 with the same siteport causes vivado to fail routing

mtdudek, 2021-07-15:
Today I stumbled on such issue.
After opening DCP created with RapidWright that uses physical netlist generated with nextpnr,
I tried to reroute netlist as delay values for some of the nets where approximated.
To do so I fixed all routes, so that design wouldn't change, then I started router.
It failed with report saying that there are conflicting nodes.

I investigated why this happened and came to conclusion.
If net uses the same siteport for pseudo PIP and to access LUT5, Vivado doesn't sees it as same net and says there are conflicting nodes. I suspect that when Vivado unrolls pseudo PIP it has 2 branches in routing tree and both use the same siteport, as they are in 2 different tree branches it thinks that this resource is needed 2 time while it is needed only once.

My suggestion would be to when creating DCP file, RW would unroll pseudo PIPs and check if there is another branch using the same siteport, and if so, fix it.

To reproduce this issue open dcp file in vivado and then run this TCL script:
```TCL
set nets [get_nets]
foreach net $nets {
	set_property is_route_fixed 1 [get_nets $net]
}

route_design -nets $nets
report_route_status
```
DCP file:
[murax.zip](https://github.com/Xilinx/RapidWright/files/6824460/murax.zip)

clavin-xlnx, 2021-07-22:
My apologies for the late response.  I am trying to understand the issue, but might need more information. 

The murax design you have attached appears to be correctly routed, all nets are reported to be fully routed with no errors per `report_route_status`.  

It sounds like you want the Vivado router to improve the routing implementation but I don't understand the motivation for fixing (locking) the routing before running the router.  When the routing is fixed, the router cannot change the route and would not be able to improve the performance, I'm not sure if this is a use case worth supporting unless someone can convince me otherwise.  

If I were attempting to improve the routing, I would simply call `route_design`.  Optionally, you could also call `lock_design -level placement` to ensure that the placement and logical portions of the design remain unchanged but I don't know if the router can change those much.

I do see that Vivado is having an issue finishing up after the call to `route_design`, `report_route_status` afterwards reports:

```
Design Route Status
                                               :      # nets :
   ------------------------------------------- : ----------- :
   # of logical nets.......................... :       13396 :
       # of nets not needing routing.......... :        9145 :
           # of internally routed nets........ :        9032 :
           # of nets with no loads............ :         113 :
       # of routable nets..................... :        4251 :
           # of nets with fixed routing....... :           2 :
           # of fully routed nets............. :        4250 :
       # of nets with routing errors.......... :           1 :
           # of nets with some unrouted pins.. :           1 :
   ------------------------------------------- : ----------- :

Nets with Routing Errors:
  GLOBAL_LOGIC1
    Unrouted Nodes (not associated with pins or ports) -- only the first 10 are listed, use -show_all to get the full list:
      CLBLL_L_X2Y23/CLBLL_L_B6
      CLBLL_L_X2Y24/CLBLL_LL_B6
      CLBLL_L_X2Y24/CLBLL_L_D6
      CLBLL_L_X2Y26/CLBLL_L_B6
      CLBLL_L_X2Y26/CLBLL_L_D6
      CLBLL_L_X2Y29/CLBLL_LL_A6
      CLBLL_L_X2Y29/CLBLL_LL_B6
      CLBLL_L_X2Y29/CLBLL_L_D6
      CLBLL_L_X2Y30/CLBLL_LL_A6
      CLBLL_L_X2Y30/CLBLL_LL_B6
```

This seems to be an artifact of a unique placement style of nextpnr, running `route_design` a second time allows the router to resolve these issues.  

If you could post the interchange files associated with the DCP you sent, that could be helpful in trying to dive deeper on a few other issues that I see, but I'm not sure what else I can offer based on your request.

mtdudek, 2021-07-23:
I'm attaching rest of capnp files.
[murax_capnp_files.zip](https://github.com/Xilinx/RapidWright/files/6866986/murax_capnp_files.zip)

I was fixing and rerouting nets, as some of them appear to have 2 subtrees. For some of these nets timing report was inaccurate, timing data was only an approximation but rerouting fixed net gave proper timing data for net.

I've checked physical netlists before and after converting to DCP. I saw that after conversion RW added stubs creating more subtrees for a net. This happens to nets that are using 2 output sitePorts in single site, like C and CMUX. I also check Vivado's CLBLL/CLBLM routing pips and I saw pseudo PIPs to route from C to CMUX.
This gave me idea that maybe Vivado net format doesn't support multiple site exists.

clavin-xlnx, 2021-07-23:
Thanks for the interchange files.  I've looked at the design and I do suspect that something is not right with some of the routes.  Even though Vivado declares the routes error free upon load, if I were to read the DCP back into RapidWright, I get several "ERROR: island/loop discovered" upon writing the DCP back out meaning there is a change taking place to the routing.  

I did just recently patch the issue you described when a net uses two site pins to drive the routing tree, this is resolved in [patch3 of 2020.2.6](https://github.com/Xilinx/RapidWright/releases/download/v2020.2.6-beta/rapidwright_api_lib-2020.2.6-patch3.zip).  However, even with the patch, there is something strange with the route that gets modified (branch annotations?) when written to the DCP.  I will continue to explore, but if you could also describe specific problematic nets that you are seeing, that would be useful.

clavin-xlnx, 2021-07-26:
One issue I have found is that at it appears one more more bi-directional PIPs from the interchange (*.phys) file is incorrectly specifying the [forward field](https://github.com/SymbiFlow/fpga-interchange-schema/blob/78abf3f30770ccc6d0e1f5dbfeaef2666f55acf6/interchange/PhysicalNetlist.capnp#L132).  In non-bi-directional PIPs, the signal flows from wire0->wire1.  However, this is sometimes ambiguous without context for bi-directional PIPs, hence the need for the forward field.  One example I found was in net `$abc$19989$memory\\murax.system_cpu.RegFilePlugin_regFile$wren[29][0][0]$y$5187`, PIP `INT_R_X9Y28/INT_R.LV0<<->>LH0`.  The usage of this PIP is actually opposite (LH0 -> LV0).  

However, RapidWright is not without fault as the code to read in the Interchange was also not checking that field, so there are two errors here.  I have checked in an update to resolve this.  Once you have updated your design with the appropriate forward flags set, I can take another look to see if there are other issues at play.

Thanks for your patience, there was a fair amount of indirection in trying to discover what is going on here.

mtdudek, 2021-07-30:
Sorry but this week I couldn't tests it and fix nextpnr netlist output.

clavin-xlnx, 2022-02-04:
Closing for now, reopen if you are able to produce a test case.

closed 2022-02-04 by clavin-xlnx