Xilinx/RapidWright #209: OBUFDS_DUAL_BUF macro missing

gatecat, 2021-07-27:
Vivado has a `OBUFDS_DUAL_BUF` macro for pseudo-differential, non-tristate, differential outputs:

[image]

But its expansion seems to be missing from RapidWright, as the following code:

```Java
package net.gatecat.fpga;

import java.io.IOException;

import com.xilinx.rapidwright.device.*;
import com.xilinx.rapidwright.design.*;
import com.xilinx.rapidwright.edif.*;

public class MacroTest {
	public static void main(String[] args) throws IOException {
		Device d = Device.getDevice("xc7a35tcsg324-1");
		EDIFLibrary macros = Design.getMacroPrimitives(d.getSeries());
		System.out.println("OBUFTDS_DUAL_BUF: " + (macros.getCell("OBUFTDS_DUAL_BUF") != null));
		System.out.println("OBUFDS_DUAL_BUF: " + (macros.getCell("OBUFDS_DUAL_BUF") != null));
	}
}
```

prints:
```
OBUFTDS_DUAL_BUF: true
OBUFDS_DUAL_BUF: false
```

clavin-xlnx, 2021-07-27:
Indeed, the OBUFTDS_DUAL_BUF was a special case to get it inserted, looks like OBUFDS_DUAL_BUF is also in that category.  I will attempt to regenerate the files with the missing macro and post a patch and/or include it in a new release.

clavin-xlnx, 2021-07-27:
You can get a patch for the missing macro with the following commands:
```
cd $RAPIDWRIGHT_PATH
wget https://github.com/Xilinx/RapidWright/releases/download/v2020.2.7-beta/rapidwright_api_lib-2020.2.7-patch1.zip
unzip -o rapidwright_api_lib-2020.2.7-patch1.zip
```

gatecat, 2021-07-27:
Thanks! Unfortunately, when trying latest RapidWright with the patch, DCP generation fails for any design, even one not using any differential IO:

```
Exception in thread "main" java.lang.RuntimeException: cannot have empty cell insts
	at com.xilinx.rapidwright.edif.EDIFHierCellInst.<init>(EDIFHierCellInst.java:109)
	at com.xilinx.rapidwright.edif.EDIFHierCellInst.createRelative(EDIFHierCellInst.java:95)
	at com.xilinx.rapidwright.edif.EDIFCell.getAllLeafDescendants(EDIFCell.java:516)
	at com.xilinx.rapidwright.edif.EDIFCell.getAllLeafDescendants(EDIFCell.java:500)
	at com.xilinx.rapidwright.interchange.PhysNetlistReader.checkMacros(PhysNetlistReader.java:651)
	at com.xilinx.rapidwright.interchange.PhysNetlistReader.readPhysNetlist(PhysNetlistReader.java:88)
	at com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp.main(PhysicalNetlistToDcp.java:38)
```

test files (just a counter): 
[counter.zip](https://github.com/Xilinx/RapidWright/files/6887864/counter.zip)

gatecat, 2021-07-27:
With the failing check patched out, DCP generation works again and I can confirm that OBUFDS now expands correctly and passes DRC with the patch:

```patch
diff --git a/src/com/xilinx/rapidwright/edif/EDIFHierCellInst.java b/src/com/xilinx/rapidwright/edif/EDIFHierCellInst.java
index 888dc13..166fd24 100644
--- a/src/com/xilinx/rapidwright/edif/EDIFHierCellInst.java
+++ b/src/com/xilinx/rapidwright/edif/EDIFHierCellInst.java
@@ -105,9 +105,6 @@ public class EDIFHierCellInst {
      * @param cellInsts the cells to use
      */
     private EDIFHierCellInst(EDIFCellInst[] cellInsts) {
-        if (cellInsts.length == 0) {
-            throw new RuntimeException("cannot have empty cell insts");
-        }
         this.cellInsts = cellInsts;
     }
```

clavin-xlnx, 2021-07-27:
Thanks for the error. We have made a significant change in 2020.2.7 to how hierarchical names and references are made in the netlist and despite significant testing, it seems we still have missed a case.  The patch will also work for 2020.2.6 as well if you want to revert and avoid changing code, but we'll definitely get this failure resolved soon.

clavin-xlnx, 2021-07-27:
Thanks @jakobwenzel your fix works for me!

gatecat, 2021-07-27:
Thanks, that was an impressively quick fix! All seems to be working now.

closed 2021-07-27 by gatecat