Xilinx/RapidWright #214: BRAM_INT_INTERFACE Routing Problem

e7p, 2021-08-03:
I've got a weird case where it is not possible to continue routing a BRAM output net (in this case however DOAD0 on an xc7a200t). I do require a custom router for my application, so I followed the [RapidWright tutorial](/clavin-xlnx/RapidWright-binder/blob/master/BasicRouting.ipynb) to implement a simple Router.
The first three PIPs are found as represented by these nodes:
- `BRAM_L_X28Y245/BRAM_RAMB18_DOADO0`
- `BRAM_L_X28Y245/BRAM_LOGIC_OUTS_B15_2`
- `BRAM_INT_INTERFACE_L_X28Y247/INT_INTERFACE_LOGIC_OUTS_L_B15`
- `BRAM_INT_INTERFACE_L_X28Y247/INT_INTERFACE_LOGIC_OUTS_L15`

However then no more connections from this node can be found. Reproduce with the following:
```python
from com.xilinx.rapidwright.device import Device
device = Device.getDevice("xc7a200t")
tile = device.getTile("BRAM_INT_INTERFACE_L_X28Y247")
tile.getWireConnections("INT_INTERFACE_LOGIC_OUTS_L15")
```
yields to an empty list `[]`, while it should lead to this one probably:
```python
tile = device.getTile("INT_L_X28Y247")
tile.getWireConnections("LOGIC_OUTS_L15")
```
correctly showing the next connection options `[INT_L_X28Y247/WW4BEG3, ...]`.

Is this a bug or some error in my connection database? I have no good clue what to look for.

clavin-xlnx, 2021-08-03:
This might be due to an older installation of RapidWright.  Be sure to update to the latest release (2020.2.7) and replace the `RapidWright/data` and `RapidWright/jars` directories with those release in the most recent version.  If you are on Linux, you can run the Makefile target `cd $RAPIDWRIGHT_PATH && make update_jars` to do this automatically.

I ran your example code and the return value of `tile.getWireConnections("INT_INTERFACE_LOGIC_OUTS_L15")` is not an empty array but the wire you mention `[INT_L_X28Y247/LOGIC_OUTS_L15]`.

e7p, 2021-08-24:
Thank you for the hint with regard to update RapidWright. After updating the data / jars indeed it was possible to find the respective connection.

closed 2021-08-24 by e7p