Xilinx/RapidWright #216: Inconsistencies around cell parameters in device resources

mkurc-ant, 2021-08-05:
I've spotted a number of inconsistencies in device resource written by RapidWright around cell parameters that control pin inversions. The following examples are for `IS_CLKINSEL_INVERTED` of `PLLE2_ADV` but I've seen similar issues with other `IS_<something>_INVERTED` parameters.

In a device resources file (converted to JSON for convenience) the `IS_CLKINSEL_INVERTED` parameter is defined as string and has the default value of `1'b0` which is more of "verilog binary":
```
{
  "name": 82785,
  "format": "string",
  "default": {
    "key": "IS_CLKINSEL_INVERTED",
    "textValue": "1'b0"
  }
},
```

Then in the section defining invertable cell pins it is said that it should be equal to either `TRUE` or `FALSE`:
```
{
  "cellPin": "CLKINSEL",
  "notInverting": {
    "parameter": {
      "key": "IS_CLKINSEL_INVERTED",
      "textValue": "FALSE"
    }
  },
  "inverting": {
    "parameter": {
      "key": "IS_CLKINSEL_INVERTED",
      "textValue": "TRUE"
    }
  }
}
```

And Vivado seems to be ok when the parameter is set to `1'b0` in a DCP built from a design routed by `nextpnr`.

clavin-xlnx, 2021-08-05:
Oddly, Vivado uses "strings" to capture Boolean property values.  Because of this, it accepts different string values to interpret them  and possibly different people/teams have used different conventions due to the flexibility.  This might need to be captured in the Interchange as a rule of thumb or other constraint on properties.  As it stands, this is technically not incorrect.

mkurc-ant, 2021-08-24:
Since Vivado accepts `TRUE` and `FALSE` for boolean parameters would it be possible to define them as boolean in device resources? This would not require any changes to the schema but will provide all the necessary hints to other tools than Vivado.

clavin-xlnx, 2022-02-04:
@acomodi - does this PR (https://github.com/chipsalliance/python-fpga-interchange/pull/121) resolve this issue?

acomodi, 2022-02-07:
@clavin-xlnx I believe yes, as there is not direct way to get Vivado to output a true boolean instead of a string, so I think that https://github.com/chipsalliance/python-fpga-interchange/pull/121 should be enough for now

closed 2022-02-07 by clavin-xlnx