Xilinx/RapidWright #217: Interchange: wrong strings for macro mapping rules

acomodi, 2021-08-05:
With some recent updates, possibly after https://github.com/Xilinx/RapidWright/pull/213, new macro rules were added, but I think that there may be some issues with the naming assignement to the macro rules.

Below there is an extract from the JSON version of the xc7z010 device, corresponding to what should have been a LUTRAM, but instead the primName and macroName correspond to a tile.

```
       "primName": "INT_L_X6Y10",                                                                                                                                                                                                                                                
       "macroName": "INT_L_X6Y10",                                                                                                                                                                                                                                               
       "always": null,                                                                                                                                                                                                                                                           
       "paramMapping": [                                                                                                                                                                                                                                                         
         {                                                                                                                                                                                                                                                                       
           "primParam": "INIT",                                                                                                                                                                                                                                                  
           "instName": "DP.HIGH",                                                                                                                                                                                                                                                
           "instParam": "INIT",                                                                                                                                                                                                                                                  
           "bitSlice": [                                                                                                                                                                                                                                                         
             64,                                                                                                                                                                                                                                                                 
             65,                                                                                                                                                                                                                                                                 
             66,                                                                                                                                                                                                                                                                 
             67,                                                                                                                                                                                                                                                                 
             ...                                                                                                                                                                                                                                                           
``` 

**Environment info**
RapidWright revision: 516d2f77dc4e4f83155b34c32f0c0c3ece35b9c6
Applied patch for the OBUFDS: https://github.com/Xilinx/RapidWright/releases/download/v2020.2.7-beta/rapidwright_api_lib-2020.2.7-patch1.zip

clavin-xlnx, 2021-08-05:
I've confirmed that the `primName` and `macroName` field for entries that are not expansion exceptions but those with parameter rule propagations are not being populated.  I will submit a fix that populates those fields to the same name as the macroName is not meaningful for these exceptions.

clavin-xlnx, 2021-08-05:
I was able to re-build the Interchange device file and convert it to JSON via python-fpga-interchange and the tile names are gone:

```
$ egrep 'macroName|primName' xc7z010.device.json 
      "primName": "LUT6_2",
      "macroName": "LUT6_2",
      "primName": "OBUFDS",
      "macroName": "OBUFDS_DUAL_BUF",
      "primName": "OBUFTDS",
      "macroName": "OBUFTDS_DUAL_BUF",
      "primName": "RAM128X1D",
      "macroName": "RAM128X1D",
      "primName": "RAM128X1S",
      "macroName": "RAM128X1S",
      "primName": "RAM16X2S",
      "macroName": "RAM16X2S",
      "primName": "RAM16X4S",
      "macroName": "RAM16X4S",
      "primName": "RAM16X8S",
      "macroName": "RAM16X8S",
      "primName": "RAM256X1D",
      "macroName": "RAM256X1D",
      "primName": "RAM256X1S",
      "macroName": "RAM256X1S",
      "primName": "RAM32M",
      "macroName": "RAM32M",
      "primName": "RAM32M16",
      "macroName": "RAM32M16",
      "primName": "RAM32X1D",
      "macroName": "RAM32X1D",
      "primName": "RAM32X2S",
      "macroName": "RAM32X2S",
      "primName": "RAM32X4S",
      "macroName": "RAM32X4S",
      "primName": "RAM32X8S",
      "macroName": "RAM32X8S",
      "primName": "RAM64M",
      "macroName": "RAM64M",
      "primName": "RAM64M8",
      "macroName": "RAM64M8",
      "primName": "RAM64X1D",
      "macroName": "RAM64X1D",
      "primName": "RAM64X2S",
      "macroName": "RAM64X2S",
      "primName": "ROM128X1",
      "macroName": "ROM128X1",
      "primName": "ROM256X1",
      "macroName": "ROM256X1",

```

closed 2021-08-05 by clavin-xlnx

acomodi, 2021-08-06:
@clavin-xlnx Great, thanks for the quick fix! I can confirm this is working as well for me