Xilinx/RapidWright #228: Getting clock region from tile fails on Versal

gatecat, 2021-09-16:
Code example:
```java
package net.gatecat.fpga;

import com.xilinx.rapidwright.design.*;
import com.xilinx.rapidwright.device.*;
import java.io.IOException;

public class clock_region {
    public static void main(String[] args) throws IOException {
        Device dev = Device.getDevice("xcvc1902-vsva2197-1MP-e-S");
        System.out.println(dev.getTile("CLE_W_CORE_X13Y4").getClockRegion().getName());
    }
}
```

This should give `X1Y1`, but actually fails with a null pointer exception internal to `getClockRegion`:

```
% ./scripts/invoke_rapidwright.sh net.gatecat.fpga.clock_region
Exception in thread "main" java.lang.NullPointerException: Cannot invoke "com.xilinx.rapidwright.device.Tile.getColumn()" because "<local8>" is null
	at com.xilinx.rapidwright.device.Device.d(Unknown Source)
	at com.xilinx.rapidwright.device.Device.getClockRegionFromTile(Unknown Source)
	at com.xilinx.rapidwright.device.Tile.getClockRegion(Unknown Source)
	at net.gatecat.fpga.clock_region.main(clock_region.java:10)
```

clavin-xlnx, 2021-09-16:
Thanks @gatecat for pointing this out and the example.  It turns out there was a bug in how clock regions were being populated for Versal devices because the clock region array is not rectangular (X10Y5 is off in the top right corner, for example).  I've made a fix and uploaded a patch for the current 2021.1.0 release:

```
cd $RAPIDWRIGHT_PATH
wget https://github.com/Xilinx/RapidWright/releases/download/v2021.1.0-beta/rapidwright_api_lib-2021.1.0-patch1.zip
unzip -o rapidwright_api_lib-2021.1.0-patch1.zip
```

With the patch, I seem to get the correct answer for all tiles in all available Versal devices.  Note that there are some tiles in some devices that legitimately do not belong to a clock region.  This doesn't appear to happen in Versal, however.

gatecat, 2021-09-16:
Thanks, can confirm the patch is working for me!

closed 2021-09-16 by gatecat