Xilinx/RapidWright #244: Placing SLICEL instances on SLICEM

jakobwenzel, 2021-09-28:
I have been running into issues placing a SiteInst with SiteTypeEnum SLICEL on a Site of type SLICEM on Artix 7. The JavaDoc of `Device.getAllCompatibleSites` uses that exact case as an example:

> [...] For example, a SLICEL can be placed at all SLICEL sites AND all SLICEM sites. If the type given were SLICEL, this method would return an array of all SLICEL and SLICEM sites.

And indeed, when calling `Device.getAllCompatibleSites` with SLICEL, it returns Slices of both types.

However, when actually placing SLICEL instances on SLICEM, the design fails to load in Vivado:

```
CRITICAL WARNING: [Constraints 18-4942] Sitetype SLICEL is not a primary or alternative of sitetype of site SLICE_X2Y99. This site will not be restored.
ERROR: [Common 17-49] Internal Data Exception: HDDMProto::readMessage failed
```

Are they actually compatible?

<details>
  <summary>Example Code to Reproduce, Click to Expand</summary>
  This code writes a design with a single SLICEL SiteInst placed at a SLICEL to `slicel.dcp`. It then calls `device.getAllCompatibleSites(SiteTypeEnum.SLICEL)` and places the same SiteInst to the first returned Site of Type SLICEM. The design is then written to `slicem.dcp`.

```Java
package com.xilinx.rapidwright.design;

import com.xilinx.rapidwright.device.BEL;
import com.xilinx.rapidwright.device.Device;
import com.xilinx.rapidwright.device.Site;
import com.xilinx.rapidwright.device.SiteTypeEnum;
import com.xilinx.rapidwright.edif.EDIFCell;
import com.xilinx.rapidwright.edif.EDIFDesign;
import com.xilinx.rapidwright.edif.EDIFLibrary;
import com.xilinx.rapidwright.edif.EDIFNetlist;
import com.xilinx.rapidwright.edif.EDIFTools;

import java.util.Arrays;

public class SliceCompatibility {

    private static EDIFNetlist createEmptyNetlist(String device) {
        EDIFNetlist netlist = new EDIFNetlist("test");
        netlist.setDesign(new EDIFDesign("test"));
        final EDIFLibrary workLibrary = netlist.getWorkLibrary();
        final EDIFCell topCell = new EDIFCell(workLibrary, "myTestCell");
        netlist.getDesign().setTopCell(topCell);

        EDIFTools.ensureCorrectPartInEDIF(netlist, device);

        return netlist;
    }

    private static Design createSampleDesign(String device) {
        final EDIFNetlist netlist = createEmptyNetlist(device);
        final Design design = new Design(netlist);

        final Cell myCell = design.createCell("myCell", Unisim.FDRE);

        final Site site = design.getDevice().getAllSitesOfType(SiteTypeEnum.SLICEL)[0];
        design.createSiteInst(site);
        final BEL bel = site.getBEL("AFF");

        design.placeCell(myCell, site, bel);

        if (design.getSiteInsts().size()!=1) {
            throw new RuntimeException("not one siteInst");
        }

        return design;
    }

    public static void main(String[] args) {
        Design design = createSampleDesign("xc7a200t");
        SiteInst siteInst = design.getSiteInsts().iterator().next();

        printPlacementInfo(siteInst);
        design.writeCheckpoint("slicel.dcp");

        Site site = findACompatibleSliceM(siteInst, design.getDevice());
        siteInst.place(site);
        printPlacementInfo(siteInst);
        design.writeCheckpoint("slicem.dcp");

    }

    private static void printPlacementInfo(SiteInst siteInst) {
        System.out.println("inst is a "+ siteInst.getSiteTypeEnum()+" placed on "+ siteInst.getSite()+", a "+ siteInst.getSite().getSiteTypeEnum());
    }

    private static Site findACompatibleSliceM(SiteInst si, Device device) {
        for (Site site : device.getAllCompatibleSites(si.getSiteTypeEnum())) {
            if (site.getSiteTypeEnum()==SiteTypeEnum.SLICEM) {
                return site;
            }
        }
        throw new RuntimeException("Did not find a SLICEM");
    }
}
```
</details>

As expected, `Site.isCompatibleSiteType()` returns true when querying a SLICEM Site with SLICEL as parameter. 
However, calling a SLICEM Site's `Site.getAlternateSiteTypeEnums()` returns an empty array. I would have expected to see SLICEM as part of the returned array.

clavin-xlnx, 2021-09-29:
Yes, this seems to not be a compatible pairing.  In previous architectures, I believe this was true.  However, as you have pointed out, a SLICEL site instance is not directly compatible with a SLICEM on Series 7 (or UltraScale/UltraScale+ as well):

[image]

I will remove these compatible placement entries in the next release.

clavin-xlnx, 2021-09-29:
This should be resolved in 2021.1.1.  It is quite possible that you can place a `SLICEL` onto a `SLICEM`, but it might have to change the `SiteTypeEnum` first.  I believe in ISE, this was supported but Vivado might not accommodate that anymore because it uses a cell-based placement rather than a site-based placement.

closed 2021-10-11 by clavin-xlnx