Xilinx/RapidWright #299: Misleading Error Message on Outdated rapidwright-api-lib-*.jar

jakobwenzel, 2021-11-29:
On one RapidWright installation, I forgot to update the API lib from `rapidwright-api-lib-2021.1.2.jar` to the current `rapidwright-api-lib-2021.2.0.jar`. Trying to load any Device fails with this exception:

> java.lang.RuntimeException: Error, the current version of RAPIDWRIGHT is not compatible with the device file(s) present on this installation.  Delete the 'data' directory and replace it with rapidwright_data.zip from https://github.com/Xilinx/RapidWright/releases/latest .
> Current RAPIDWRIGHT device file version: 0.7.7, existing device file version: 0.7.8.

This error message is misleading. Redownloading the data directory will not fix this. The automatic datafiles updater worked as intended and updated the files.

The actual problem is that the API lib is too old. It would be great if RapidWright could detect this issue and print a more helpful message.

clavin-xlnx, 2021-11-29:
Thanks @jakobwenzel, we'll get this message updated to inform the user that they likely need to run `gradlew update_jars` to resolve the issue.  Probably will be included in the next release.

jakobwenzel, 2021-12-01:
I just ran into another related issue: I was on a old branch of mine that did not include the latest release. Therefore that branch's `DataVersions.java` contains checksums to download that old release's data files.

However, `gradlew update_jars` will always download the newest api lib, not the one matching the currently checked out RapidWright. Running RapidWright, the error message is the same as above, just with switched versions:

> Exception in thread "main" java.lang.RuntimeException: Error, the current version of RAPIDWRIGHT is not compatible with the device file(s) present on this installation.  Delete the 'data' directory and replace it with rapidwright_data.zip from https://github.com/Xilinx/RapidWright/releases/latest .
> Current RAPIDWRIGHT device file version: 0.7.8, existing device file version: 0.7.7.

Redownloading the data directory does not fix this issue either, because the files get downgraded again to 0.7.7 by the automatic datafiles updater.

It would be great if `gradlew update_jars` could figure out the correct version to download.

Alternatively, I think moving `DataVersions.java` to the API lib would fix the issue as well.

eddieh-xlnx, 2022-01-28:
Is this issue resolved now that https://github.com/Xilinx/RapidWright/pull/320 has been merged?

clavin-xlnx, 2022-02-04:
Yes, Gradle dependencies are fully operational now.

closed 2022-02-04 by clavin-xlnx