Xilinx/RapidWright #302: Some question Bitstream Manipulation

kangliyu1, 2021-12-02:
Hello! Seniors, do you use rapidwright for bitstream analysis? It is mentioned in the Bitstream Manipulation section of the document that this function can be realized. Are there any more specific steps in this section? Hope someone can give some pointers, thank you very much

clavin-xlnx, 2021-12-02:
Bitstream analysis at the coarse-grained level can be achieved, although we have not undertaken any specific projects around this topic.  This is mostly useful for reliability analysis of a design (finding the number of sensitive bits to upsets from radiation, etc).  Perhaps if you can share more about the kind of analysis you'd like to perform, we could provide more direction.

kangliyu1, 2021-12-06:
> Bitstream analysis at the coarse-grained level can be achieved, although we have not undertaken any specific projects around this topic. This is mostly useful for reliability analysis of a design [...]

Thank you for your reply. Let me give you an example. Suppose I want to lay out 8 LUTs in a CLB module of ROW1 of the xc7s50fgga484 chip, and initialize them to 1 (the program and constraints are shown below), and use the Bitstream Manipulation function How to locate the corresponding frame
code:
```
module bram(
output wire O1,
output wire O2,
output wire O3,
output wire O4,
output wire O5,
output wire O6,
output wire O7,
output wire O8,
input wire I0,
input wire I1,
input wire I2,
input wire I3,
input wire I4,
input wire I5
    );
LUT6 #(.INIT(64'h1111111111111111))
    LUT6_inst1(
        .O(O1),
        .I0(I0),
        .I1(I1),
        .I2(I2),
        .I3(I3),
        .I4(I4),
        .I5(I5)       
    );
    
    LUT6 #(.INIT(64'h1111111111111111))
    LUT6_inst2(
        .O(O2),
        .I0(I0),
        .I1(I1),
        .I2(I2),
        .I3(I3),
        .I4(I4),
        .I5(I5)
     
    );
    LUT6 #(.INIT(64'h1111111111111111))
    LUT6_inst3(
        .O(O3),
        .I0(I0),
        .I1(I1),
        .I2(I2),
        .I3(I3),
        .I4(I4),
        .I5(I5)
            
    );
    LUT6 #(.INIT(64'h1111111111111111))
    LUT6_inst4(
        .O(O4),
        .I0(I0),
        .I1(I1),
        .I2(I2),
        .I3(I3),
        .I4(I4),
        .I5(I5)
            
    );
    LUT6 #(.INIT(64'h1111111111111111))
    LUT6_inst5(
        .O(O5),
        .I0(I0),
        .I1(I1),
        .I2(I2),
        .I3(I3),
        .I4(I4),
        .I5(I5)
            
    );
    LUT6 #(.INIT(64'h1111111111111111))
    LUT6_inst6(
        .O(O6),
        .I0(I0),
        .I1(I1),
        .I2(I2),
        .I3(I3),
        .I4(I4),
        .I5(I5)
            
    );
    LUT6 #(.INIT(64'h1111111111111111))
    LUT6_inst7(
        .O(O7),
        .I0(I0),
        .I1(I1),
        .I2(I2),
        .I3(I3),
        .I4(I4),
        .I5(I5)
            
    );
    LUT6 #(.INIT(64'h1111111111111111))
    LUT6_inst8(
        .O(O8),
        .I0(I0),
        .I1(I1),
        .I2(I2),
        .I3(I3),
        .I4(I4),
        .I5(I5)
            
    );
    

endmodule
```

constrains:
```
set_property BEL D6LUT [get_cells LUT6_inst1]
set_property LOC SLICE_X0Y100 [get_cells LUT6_inst1]
set_property BEL C6LUT [get_cells LUT6_inst2]
set_property LOC SLICE_X0Y100 [get_cells LUT6_inst2]
set_property BEL B6LUT [get_cells LUT6_inst3]
set_property LOC SLICE_X0Y100 [get_cells LUT6_inst3]
set_property BEL A6LUT [get_cells LUT6_inst4]
set_property LOC SLICE_X0Y100 [get_cells LUT6_inst4]
set_property BEL D6LUT [get_cells LUT6_inst5]
set_property LOC SLICE_X1Y100 [get_cells LUT6_inst5]
set_property BEL C6LUT [get_cells LUT6_inst6]
set_property LOC SLICE_X1Y100 [get_cells LUT6_inst6]
set_property BEL B6LUT [get_cells LUT6_inst7]
set_property LOC SLICE_X1Y100 [get_cells LUT6_inst7]
set_property BEL A6LUT [get_cells LUT6_inst8]
set_property LOC SLICE_X1Y100 [get_cells LUT6_inst8]

```
[image]

How can I locate the corresponding part of the content in the generated RBT file?

clavin-xlnx, 2021-12-07:
To find the configuration block (a collection of frames) that correspond to a particular cell or Tile, you can follow the example given here: http://www.rapidwright.io/docs/Bitstream_Manipulation.html#example-usages-find-and-print-the-frames-of-a-placed-cell

RapidWright is not able to pin point finer granularity.  However, if you are willing to switch from `LUT6` to a memory-style primitive such as `RAM64X1S`, those bits are documented and can be updated.  Another option is the `CFGLUT5` primitive which also the user to update configuration bits at runtime.

closed 2021-12-15 by kangliyu1