Xilinx/RapidWright #310: Some question Bitstream Manipulation

kangliyu1, 2021-12-15:
Hello! I want to print the frame of the CMT or PLL module of the XC7S50FGGA484-1 chip when using the print frame function of Bitstream Manipulation, but it can’t be printed. What is the problem? There is also a recent analysis of the composition of the bitstream of the XC7S50FGGA484-1 chip, and found that the frame number of each module of the chip (eg: BRAM frame is 30 frames in the circuit, but what is printed in the Bitstream Manipulation function is 28 frames) Not sure, can any seniors give me some advice? Thank you very much! !

clavin-xlnx, 2021-12-15:
It is true that RapidWright can print bitstream frames, but it does not have the capability to identify specific frames for a given primitive unless they are already documented in the logic location file.  The best you can do is print a range of frames the correspond to the tile to which a primitive belongs.  

It is difficult to diagnose and resolve the problem you are seeing based on your current description.  Perhaps you could post example code and a design/bitstream the reproduces the issue you are seeing? 

I'm not sure what you mean by "There is also a recent analysis of the composition of the bitstream" that seems to conflict with the current model.  Can you point us to this "analysis"?

kangliyu1, 2021-12-16:
> It is true that RapidWright can print bitstream frames, but it does not have the capability to identify specific frames for a given primitive unless they are already documented in the logic location [...]

@clavin-xlnx 
Hello! What I want to know is the size and position of each module of xC7S50FGGA484-1 chip in the generated RBT file after Generate Bitstream (for example: I used the Bitstream Manipulation function to Print a BRAM that I had laid out, as shown in Figure 1, and then used the "Find and Print the Frames of a Placed Cell" function to Print the frame. And then in the generated RBT file location), but after the experiment found some number
fig:1
[image]
fig:print
[image]

clavin-xlnx, 2021-12-16:
Sorry, without code and an example design, it is difficult to reproduce the problem and understand what you are trying to accomplish.  RapidWright does not read/generate RBT designs, so I'm not sure how to help there.

clavin-xlnx, 2022-02-04:
Closing for now, please reopen if further questions arise.

closed 2022-02-04 by clavin-xlnx