Xilinx/RapidWright #311: getting the physical Net from an EDIFPortInst

GBrics, 2021-12-15:
I need to get all the physical nets connected to my Cells but the only thing I can reach is the logicial Nets (EDIFNet) by using : 

```
for(EDIFPortInst port: cell.getEDIFCellInst().getPortInsts()) {
    System.out.println(port.getNet());
}
```

I tried to create a Map linking the EDIFNet to the corresponding Net using : 
```
Map<EDIFNet, Net> logicNetToNetMap = new HashMap<>();
for(Net n : nets) {
    if(n.getLogicalNet()!=null) {
        logicNetToNetMap.put(n.getLogicalNet(),n);
    }
}
```
But as a Net can be composed of several EDIFNet, the map doesn't not contains most of the EDIFNet in my design.

Is there any solution to reach my goal ? 

Thank you in advance.

clavin-xlnx, 2021-12-16:
One of the key points to keep in mind when trying to correlate logical netlist constructs with the physical netlist, is that physical netlist objects (`Cell` and `Net`) use hierarchical names.  By construction, the logical netlist uses `EDIFCellInst`, `EDIFNet` and `EDIFPortInst` which are not hierarchical.  To help translate between the two, RapidWright provides a set of hierarchical netlist objects such as `EDIFHierCellInst`, `EDIFHierNet` and `EDIFHierPortInst`.  

Having said that, there is actually a shortcut API so that you don't actually need to create a `EDIFHier*` object.  To get the physical nets from the pins of a cell you can use the API in `EDIFNetlist.getPhysicalNetFromPin(String hierParentName, EDIFPortInst portInst, Design design)`.  Therefore, if you have a `Cell` object, you could potentially get all the physical nets from its pins by using this code:

```
    public static Set<Net> getPhysicalNetsFromCell(Cell c, Design d) {
        Set<Net> nets = new HashSet<>();
        EDIFNetlist n = d.getNetlist();
        String hierParentName = c.getParentHierarchicalInstName();
        for(EDIFPortInst portInst  : c.getEDIFCellInst().getPortInsts()) {
            Net net = n.getPhysicalNetFromPin(hierParentName, portInst, d);
            if(net != null) nets.add(net);
        }
        return nets;
    }
```

GBrics, 2021-12-21:
Indeed, it works perfectly.
Thanks for the tips !

closed 2021-12-21 by GBrics