Xilinx/RapidWright #325: Wrong SiteInst location

GBrics, 2022-01-17:
Hi, 

I have been experiencing some issues when trying to get the ``SiteInst`` of one my Design ``Cell``. 

```
public static void main(String[] args) throws Exception {		
  design = Design.readCheckpoint("D:\\myfolder\\top_adder_tester_routed.dcp"); 
  myDevice = design.getDevice();
  design.flattenDesign();	
  Collection<Cell> cells = design.getCells();
  
  Cell cell = getCell(cells, "o_s[15]_INST_0_i_11"); 
  System.out.println(cell.getSiteInst());
}

public static Cell getCell(Collection<Cell> cells, String CellName) {
  Cell out0 = null;
  for(Cell c : cells)
  {
    if(c.getEDIFCellInst() != null) {
      String cell_name = c.getEDIFCellInst().getName();
      if (cell_name.equals(CellName)) {
        out0=c;
      }
    }
  }
  return out0;
}
```

This piece of code returns :
```
SiteInst(name="SLICE_X2Y134", type="SLICEM", site="SLICE_X2Y134")
```

The issue is that, according to Vivado (Device view), the cell is not located on ``SLICE_X2Y134`` but on ``SLICE_X12Y135``.

Attached is the checkpoint file:
[top_adder_tester_routed.zip](https://github.com/Xilinx/RapidWright/files/7882581/top_adder_tester_routed.zip)

Can you help me with that please?

eddieh-xlnx, 2022-01-17:
Hi @GBrics. I've just opened up your DCP in Vivado 2021.2 and I can see that there are two cells with that name.
On the Vivado Tcl console:
```
get_cells -hier -filter {NAME =~ "*/o_s[15]_INST_0_i_11"}
mult_comparator/mult0/o_s[15]_INST_0_i_11 mult_comparator/mult1/o_s[15]_INST_0_i_11
```
and they are located at:
```
get_property LOC [get_cells -hier -filter {NAME =~ "*/o_s[15]_INST_0_i_11"}]
SLICE_X12Y135 SLICE_X2Y134
```
It looks like your code only returns the last match?

GBrics, 2022-01-17:
Indeed, this was my mistake. I forgot that this block was instantiated twice in my design.
As my ``getCell`` function was searching a matching cell by looking at the ``EDIFCellInst`` name, it was not considering the hierarchy of the instance.  

I was just looking at the wrong cell, thanks for pointing this out.

closed 2022-01-17 by GBrics