Xilinx/RapidWright #326: Interchange: missing cell to bel pin mapping for some cells

acomodi, 2022-01-17:
There are some primitive cells which lack a complete cell to bel mapping. The primitives I have checked are `IDDR`, `ODDR` and `IDDR_2CLK`, but this does not exclude that there are others.

Basically, an absence of the cell to bel mapping makes it impossible to correctly read in the architecture database description, causing errors when using these primitives when place and routing using the interchange support in nextpnr and VPR.

E.g. the IDDR clock has the following IO definition (from the unisim lib):

```
module IDDR (Q1, Q2, C, CE, D, R, S);
```

In this case, the cell to bel mapping for the R and S signals is absent.

Given that the BEL corresponding to the `IDDR` cell does is provided with only a single SR pin, it might be possible that the cell to bel mapping cannot be correctly exported.

gatecat, 2022-01-17:
From memory, only one of them can be used (and becomes mapped to SR) at once. This is probably something that either needs to be modelled in the interchange format or handled beforehand in a Yosys techmap.

clavin-xlnx, 2022-01-20:
@acomodi I've added some missing entries (including `IDDR`, `ODDR` and `IDDR_2CLK`) that should include pin mappings for the missing pins.  They are based on setting particular parameters on the cells.  Try the update from #328 and let us know.

clavin-xlnx, 2022-02-04:
@acomodi was able to test in nextpnr and found that update was adequate.  Please reopen if any other issues occur.

closed 2022-02-04 by clavin-xlnx