Xilinx/RapidWright #327: same EDIFCellInst object when instanciating the same block twice

GBrics, 2022-01-18:
I have the same module "mult" instantiated twice in my design (mult0 & mult1).

I can see that the ``Cell`` objects are different between the two instantiations. However, the corresponding ``EDIFCellInst`` obtained with ``.getEDIFCellInst()`` method are the same object.

For example: 

```
public static void main(String[] args) throws Exception {		
  design = Design.readCheckpoint("D:\\myfolder\\top_adder_tester_routed.dcp"); 
  myDevice = design.getDevice();
  design.flattenDesign();	
  Collection<Cell> cells = design.getCells();
  
  Cell cell1 = getCell(cells, "mult_comparator/mult1/o_s[15]_INST_0_i_11"); 
  Cell cell2 = getCell(cells, "mult_comparator/mult0/o_s[15]_INST_0_i_11");
    
  EDIFCellInst edifCell1 = cell1.getEDIFCellInst();
  EDIFCellInst edifCell2 = cell2.getEDIFCellInst();	
  
  System.out.println(cell1);
  System.out.println(cell2);
  System.out.println(cell1.hashCode());
  System.out.println(cell2.hashCode());
  
  System.out.println(edifCell1.hashCode());
  System.out.println(edifCell2.hashCode());
}

public static Cell getCell(Collection<Cell> cells, String CellName) {
  Cell out0 = null;
  for(Cell c : cells)
  {
    if(c.getName().contains(CellName)) {
      out0 = c;
    }
  }
  return out0;
}
```
With this piece of code, I obtain : 
```
mult_comparator/mult1/o_s[15]_INST_0_i_11(BEL: CARRY4)
mult_comparator/mult0/o_s[15]_INST_0_i_11(BEL: CARRY4)
1516156071
1304805158
-1481790717
-1481790717
```
We can see that the two ``EDIFCellInt`` have the same hashcode.

Is this the intended behavior ? If yes, is there any way to obtain distinct ``EDIFCellInst`` objects when the same module is instanciated several times on the design ? 

Thank you in advance.

checkpoint file below:
[top_adder_tester_routed.zip](https://github.com/Xilinx/RapidWright/files/7890202/top_adder_tester_routed.zip)

clavin-xlnx, 2022-01-20:
You bring up a good point and we are having discussions on exactly what it should mean for an `EDIFCellInst` and related objects to be equal.  This was also brought up in Issue #42.  

To give you some background, in this instance, EDIFCellInst do have the same `hashCode()` value, but that is because they have the same name.  Many `EDIF*` objects inherit from `EDIFName` whose `equals()` and `hashCode()` methods are defined by the `name` or `edifName`.  This obviously can lead to some problems in several contexts.  

If you were to add the following code to your program above:
```
System.out.println(edifCell1 == edifCell2);
System.out.println(edifCell1.getParentCell() + " ==? " + edifCell2.getParentCell());
```
it would output:
```
false
simple_multiplier ==? simple_multiplier__1
```

The first line checks if the `EDIFCellInst` objects themselves are the same (checks if the reference is equal).  In this case they are not, because they exist in different cells as shown by the second line.  The parent cell (or cells to which these instances belong) are defined as different cells.  Vivado actually will generally create new cells for each instance when it synthesizes a design.  This makes netlist manipulation easier, but can cost additional memory.  So even though RapidWright tells you that the `EDIFCellInst` objects are equal and have the same hashCode, they are actually different objects. 

We are working on a solution to help remedy this scenario, but hopefully for now, you can just be mindful and use the `==` comparator instead of `equals()`.

GBrics, 2022-02-03:
OK, I understand your point.
Maybe a solution would be to use both the cell name and the parent cell name concatenated to generate the hashCode(). It's just a suggestion (I do not know if you have access to this part of the code).
Anyway, thanks for the help!

closed 2022-02-03 by GBrics