Xilinx/RapidWright #342: Dependency Version Mismatch: ZeroMQ - jupyter-kernel-jsr223

Mrcrypt, 2022-02-18:
The dependency ZeroMQ has been updated from 0.3.6 in RapidWright version 2021.1.0 to 0.5.3 in RapidWright 2021.2.1.
This causes an Issue with the jupyter-kernel-jsr223 depenendy, because its compiled aganst the older version of ZeroMQ. The "bind" method has changed in ZeroMQ and that casuses a crash if you try to use jupyter notebook with the newest RapidWright version.

A fix would be to set the ZeroMQ version to 0.3.6 again, as the jupyter-kernel-jsr223 hasnt been updated in years.

clavin-xlnx, 2022-02-19:
I've released an update (see pull request #344) to the jupyter-kernel-jsr223 (1.0.0 to 1.0.1) which includes an update so that it compiles and depends on JeroMQ 0.5.2.  Please give it a try, I will do more extensive testing next week.

eddieh-xlnx, 2022-03-14:
#344 merged.

closed 2022-03-14 by eddieh-xlnx