Xilinx/RapidWright #351: Adding an API to insert a buffer cell (LUT1 or FF) on a module's ports

clavin-xlnx, 2022-02-26:
One useful capability would be to automatically add a buffer cell (LUT1 or FDRE) similar to the way "update_design -buffer_ports" operates in Vivado.  This would help with integrating multiple designs together into a single one.

athanas, 2022-03-04:
There seems to be an issue making a connection after importing a DCP.   The basic steps to cause the issue:
1. design = Design.readCheckpoint(...)
2. a_net = design.createNet(...)
3. a_net.connect(...)
4. design.routeSites()
After (4) is executed, a null pointer exception is raised from EDIFHierPortInst().  The ugly work-around is to comment out the connect, identify the net endpoints, import the DCP back into Vivado, and use the Vivado router.

I am running Vivado 2020.2 with AR76191 patch applied.

clavin-xlnx, 2022-03-04:
Thanks for bringing this up.  We do have other code that follows this pattern without issues, so it might be hard to replicate.  Would it be possible for you to share the DCP and/or code that reproduces the error?  If not, could you share the stack trace that it produced from the exception?

athanas, 2022-03-04:
Dr. Lavin, yes I can, but until then, my DCP consists simply of one inverter using an input pin and an output pin (I tried to reduce this to the minimum).  The error occurs even when the RW added net does not interact with the DCP nets.  It does not occur when readCheckpoint() is not called.

athanas, 2022-03-04:
Attached you'll find the zip containing the DCP and EDIF
[nope.zip](https://github.com/Xilinx/RapidWright/files/8182111/nope.zip)
.

clavin-xlnx, 2022-03-04:
Thanks for the DCP, but I am sorry, I don't seem to be able to infer the use case or reproduce the error.  Could you change this code such that it fails for you?  
```
import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Net;
import com.xilinx.rapidwright.design.Unisim;
import com.xilinx.rapidwright.device.Device;

/**
 * Created on: Mar 3, 2022
 */
public class Debug {

    public static void main(String[] args) {
        System.out.println(Device.FRAMEWORK_NAME_AND_VERSION);
        Design d = Design.readCheckpoint("nope.dcp");

        Cell cell = d.createAndPlaceCell("self_feed_ff", Unisim.FDRE, "SLICE_X48Y147/AFF");
        Net net = d.createNet("feedback_net");
        net.connect(cell, "Q");
        net.connect(cell, "D");
        
        d.routeSites();
        
        d.writeCheckpoint("yep.dcp");
    }
}
```
The output I get is:
```
RapidWright v2021.2.1
 device v0.7.9
==============================================================================
==                          Reading DCP: nope.dcp                           ==
==============================================================================
 XML Parse & Device Load:     1.052s
              EDIF Parse:     0.371s
        Read XDEF Header:     0.044s
        Read XDEF Caches:     0.055s
     Read XDEF Placement:     0.160s
       Read XDEF Routing:     0.050s
------------------------------------------------------------------------------
         [No GC] *Total*:     1.732s
==============================================================================
==                           Writing DCP: yep.dcp                           ==
==============================================================================
            (Write EDIF):            (    0.027s)
    (Write XDEF Routing):            (    0.436s)
     (Write XDEF Header):            (    0.003s)
  (Write XDEF Placement):            (    0.053s)
(Write XDEF Caches (1/3)):            (    0.002s)
(Write XDEF Caches (2/3)):            (    0.000s)
(Write XDEF Caches (3/3)):            (    0.007s)
 (Write XDEF Finalizing):            (    0.004s)
            (Write XDEF):            (    0.555s)
       Write EDIF & XDEF:     0.600s
               Write XDC:     0.015s
------------------------------------------------------------------------------
         [No GC] *Total*:     0.614s

```
The logical netlist view in Vivado:
[image]
The physical view of the added flop (and routed site):
[image]

athanas, 2022-03-04:
Here is the boiled-down code:
# For Python bindings for RW:
import rapidwright

# Import RapidWright classes
from com.xilinx.rapidwright.design import Cell
from com.xilinx.rapidwright.design import Design
from com.xilinx.rapidwright.design import Net
from com.xilinx.rapidwright.design import PinType
from com.xilinx.rapidwright.design import Unisim
from com.xilinx.rapidwright.device import Device, TileTypeEnum
from com.xilinx.rapidwright.router import Router
from com.xilinx.rapidwright.util   import FileTools, Utils
import os

# PART: xczu3eg-sbva484

# Open design DCP
design = Design.readCheckpoint("../nope.dcp","../nope.edf")

lut = design.createAndPlaceCell("lut", Unisim.AND2, "SLICE_X13Y42/A6LUT")

sIn = design.getNet("signal_IBUF_inst/O")
sIn.connect(lut,"I0")

# Route intra-site connections
design.routeSites()

# Route inter-site connections
Router(design).routeDesign()

# Write out the placed and routed DCP
design.writeCheckpoint("mod.dcp")

===============================================
Here is the error:
==============================================================================
==                          Reading DCP: nope.dcp                           ==
==============================================================================
Loading device from file xczu3eg-sbva484-1-e
 XML Parse & Device Load:     1.271s
              EDIF Parse:     0.313s
        Read XDEF Header:     0.020s
        Read XDEF Caches:     0.010s
     Read XDEF Placement:     0.026s
       Read XDEF Routing:     0.044s
------------------------------------------------------------------------------
         [No GC] *Total*:     1.684s
Traceback (most recent call last):
  File "com.xilinx.rapidwright.design.Design.java", line -1, in com.xilinx.rapidwright.design.Design.routeSites
  File "com.xilinx.rapidwright.design.SiteInst.java", line -1, in com.xilinx.rapidwright.design.SiteInst.routeSite
  File "EDIFNetlist.java", line 921, in com.xilinx.rapidwright.edif.EDIFNetlist.getPhysicalNetFromPin
  File "EDIFNetlist.java", line 931, in com.xilinx.rapidwright.edif.EDIFNetlist.getPhysicalNetFromPin
Exception: Java Exception

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "problem.py", line 26, in <module>
    design.routeSites()
java.lang.NullPointerException: java.lang.NullPointerException

clavin-xlnx, 2022-03-04:
Perfect, I was able to reproduce the issue.  I can see there is a limitation with `Net.connect()` that is exposed by this example.  To work around the issue, you can replace the code:
```
sIn.connect(lut,"I0") 
```
with this less attractive work-around:
```
from lang.util import ArrayList
#...
design.getTopEDIFCell().getNet("signal_IBUF").createPortInst("I0", lut)
lutInput = sIn.createPin("A5", lut.getSiteInst())
pinsToRoute = ArrayList()
pinsToRoute.add(lutInput)
```
And since you are routing a partially routed design, you'll want to invoke the router this way instead:
```
Router(design).routePinsReEntrant(pinsToRoute, true)
```

Now, what is going wrong here:  

The issue stems from the fact that the target net to connect to `signal_IBUF_inst/O` is inside a hierarchical cell:
[image]
Unfortunately, `Net.connect()` was not written to handle hierarchical scenarios (bug filed #355).  Thus, the work-around has to create the logical connection (`EDIFPortInst`) and physical connection (`SitePinInst`) manually.  

Also, just FYI, we also have a new, more robust router called `RWRoute` which handles congestion (`Router` does not) and is optionally timing-driven (see [tutorials](http://www.rapidwright.io/docs/Tutorials.html) for more information).

clavin-xlnx, 2025-05-13:
Closing as this appears resolved.  Please re-open or file a new issue if there are additional concerns.

closed 2025-05-13 by clavin-xlnx