Xilinx/RapidWright #360: Interchange: missing cell bel mappings for US+

acomodi, 2022-03-07:
While trying to add tests in the [fpga-interchange-tests](https://github.com/chipsalliance/fpga-interchange-tests/pull/98) suite, nextpnr errors out as it reports that some cells are not placeable.

This seems to be corresponding to lacking maps in the `CellBelMapping` for some cells. In particular, the ones tested, and missing are:

- PLLE3_ADV
- PLLE3_BASE
- PLLE4_BASE
- MMCME3_ADV
- MMCME3_BASE
- MMCME4_BASE

It is not excluded that there are more maps missing.

clavin-xlnx, 2022-03-07:
It appears that these cell types are not natively supported on the devices in question.  Pin mappings are not provided as Vivado will generally retarget a primitive cell of these types to the supported ones.  In #205, there was a discussion on the "retargeted" primitives and it was concluded that this should be resolved by the synthesis tool.  For reference, on an xczu3eg, running the following command: 
```
create_cell -reference PLLE3_ADV PLLE3_ADV_inst
```
Gives the following message in Vivado and generates a `PLLE4_ADV` instead of a `PLLE3_ADV`
```
WARNING: [Coretcl 2-1024] Primitives of type 'PLLE3_ADV' are not supported by the current part and have been retargeted to 'PLLE4_ADV'.
```

An exhaustive search on a xczu7ev yields these messages from Vivado:
```
WARNING: [Coretcl 2-1024] Primitives of type 'DCM_ADV' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'DCM_SP' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'MMCME2_ADV' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'MMCME2_BASE' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'MMCME3_ADV' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'MMCME3_BASE' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'MMCME4_BASE' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'PLLE2_ADV' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'PLLE2_BASE' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'PLLE3_ADV' are not supported by the current part and have been retargeted to 'PLLE4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'PLLE3_BASE' are not supported by the current part and have been retargeted to 'PLLE4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'PLLE4_BASE' are not supported by the current part and have been retargeted to 'PLLE4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'CLKDLL' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'CLKDLLE' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'CLKDLLHF' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'DCM' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'DCM_BASE' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
WARNING: [Coretcl 2-1024] Primitives of type 'DCM_PS' are not supported by the current part and have been retargeted to 'MMCME4_ADV'.
```

How difficult is it to push these "retargeted" primitives for UltraScale+ back to Yosys?  Unfortunately, there is a fair amount of work on the RapidWright side to provide this info for unsupported primitives.

acomodi, 2022-03-08:
@clavin-xlnx Thanks for the clarification. Currently we are handling retargeting by adding a custom techmap file to hand out to yosys, so, if needed, we can adopt the same mechanism here and potentially integrate it within upstream yosys so that the retargeting won't require extra steps to be performed out of the main synth and PnR tools.

closed 2022-03-08 by acomodi