Xilinx/RapidWright #367: some question

kangliyu1, 2022-03-14:
Hello! In the process of using rapidwright, the following error appeared. This problem has troubled me for a long time. I hope someone can give some advice. Many thanks! !

[image]
[image]

clavin-xlnx, 2022-03-14:
The error seems to be occurring during the expansion of macros when loading the netlist.  Certain macros get populated upon load in RapidWright to mimic the behavior in Vivado.  In this case, it can't seem to find the definition for OBUFTDS.  It is difficult to know the cause.  Would you be able to provide the EDF file as an example test case?

kangliyu1, 2022-03-15:
抄送: "さ挚さ挚 ***@***.******@***.***&gt;;

 

 

从QQ邮箱发来的超大附件

top.bit (2.09M, 2022年03月15日 09:16 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=5936646220b8e9cf57c493201735051f060f0551060006044807075403180300510249030401021d06015d070853060504030707313237440a464a005841370d&code=e6db1570

top.edf (24.28M, 2022年03月15日 09:16 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=086338622d4e499b0691cf204537004a0d5a0a500751565d19010d06001a06570652155a0556054802515d035002545605520c07633032115b13160707513258&code=4c8bc72e

top.rbt (17.77M, 2022年03月15日 09:18 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=096633307c87d6cc0794c4721736071d545407530806540118030055041b01030d021e525006071f50500b080605540407020406313135465a161d425342350f&code=5f301652

top_routed.dcp (8.03M, 2022年03月15日 09:19 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=2731656392e91cc956c3922145655119050753520503510349065357534857575006485b5454511b5d505d555554000605535d00636b63420b413a110c101753001f010013655e&code=d1eccec6

截图1.png (211.8K, 2022年03月15日 09:19 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=5b35616337d7989d57c796211166564d5d01025103510751480c0306064b505a50514c025256554f530700070203545607500201376f64dfbdf8dd5219160a056508&code=e5ac7fdb

截图2.png (189.7K, 2022年03月15日 09:19 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=076138627a6bcbca0b93cf201562041a0a550e000054525714050e5a524f02540052150056030e185a070c00505b03020c515951336b3688e1ac84501d125852395c&code=9a8b3b65

clavin-xlnx, 2022-03-15:
Sorry, I am unable to download those files.  Could you attach them using GitHub?  

See https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/attaching-files for directions.

clavin-xlnx, 2022-10-21:
Closing due to inactivity.

closed 2022-10-21 by clavin-xlnx