Xilinx/RapidWright #388: RapidWright should track in the netlist if macros have been expanded or collapsed

clavin-xlnx, 2022-04-15:
When Vivado reads a netlist (EDIF file, for example) it will expand the macros in it before returning control to the user.  RapidWright emulates this behavior by expanding upon read of a DCP and collapsing upon write of a DCP.  However, if a DCP is written as an intermediate step, the netlist in memory stays collapsed and performing other netlist operations may fail afterwards (such as `DesignTools.copyImplementation()`).  There should be a flag added to the netlist to track the current state of the macros so that these methods, or the user can query and expand the netlist appropriately.

pongstorn-xilinx, 2022-04-16:
What is the API to expand the macro?

clavin-xlnx, 2022-04-18:
> What is the API to expand the macro?

`EDIFNetlist.expandMacroUnisims(Series series)`