Xilinx/RapidWright #419: [Versal] [placeCell]Site type DSP58_PRIMARY not supported for cell type DSP_XXX

nqdtan, 2022-05-13:
Hi,

I'm encountering placement issue related to the Versal DSP BELs. Please find the code for reproducing the issue as follows. (also the DCP and EDIF files in the attached zip file.)

```java
package com.xilinx.rapidwright.examples;
import java.io.FileNotFoundException;

import com.xilinx.rapidwright.design.*;
import com.xilinx.rapidwright.device.*;

public class TestPlaceDSPVersal {
  public static void main(String[] args) throws FileNotFoundException {
    Design design1 = Design.readCheckpoint("add_placed.dcp");
    Design design2 = new Design("new_design", design1.getDevice().getName());
    for (Cell c : design1.getCells()) {
      design2.copyCell(c, c.getName());
    }
  }
}
```
**RapidWright version:** 2022.1 (beta release; it seems that the code at master somehow crashes when reading my DCP.)

**Vivado version (for the DCP):** 2022.1

**Error log**

> Exception in thread "main" java.lang.RuntimeException: ERROR: Site type DSP58_PRIMARY not supported for cell type DSP_PREADD58
	at com.xilinx.rapidwright.design.i.a(Unknown Source)
	at com.xilinx.rapidwright.design.i.a(Unknown Source)
	at com.xilinx.rapidwright.design.i.a(Unknown Source)
	at com.xilinx.rapidwright.design.Design.placeCell(Unknown Source)
	at com.xilinx.rapidwright.design.Design.copyCell(Unknown Source)
	at com.xilinx.rapidwright.examples.TestPlaceDSPVersal.main(TestPlaceDSPVersal.java:15)

The DCP only contains a single cell that does 32b addition. I think there are also other DSP cells that are currently not supported as well: DSP_ALUADD, DSP_ALUMUX, DSP_ALUREG, DSP_A_B_DATA58, DSP_C_DATA58, DSP_MULTIPLIER58, DSP_M_DATA58, DSP_OUTPUT58, DSP_PATDET, DSP_PREADD58, DSP_PREADD_DATA58, DSP_SRCMX_OPTINV.

Thanks!
[src.zip](https://github.com/Xilinx/RapidWright/files/8683972/src.zip)

clavin-xlnx, 2022-05-13:
Thanks for the issue and example code/design, always helpful to have those!  

First, there is an issue with reading 2022.1 DCPs in the latest (2022.1.0) release which we have resolved and will be available in 2022.1.1.  We've created a branch ([2022.1.1](https://github.com/Xilinx/RapidWright/tree/2022.1.1)) and PR (#420) with the candidate fix which you can get now to resolve the DCP load issue.

Second, the issue you are seeing is due to a site type mismatch.  It turns out that sites can take on different 'types'.  In the example you have provided, the `DSP_X0Y0` site can be of type `DSP58_PRIMARY`, `DSPFP` or `DSP58`, see this Vivado Site Properties window to illustrate:

[image]

In the case of the source design, the site is of type `DSP58`, however when RapidWright attempts to copy the cells into the new (and empty) design, it must create a new `SiteInst` object and `Design.copyCell()` is not smart enough to realize that the cell needs a site of type `DSP58` instead of the default `DSP58_PRIMARY`.  This is a bug and I'll work on adding those smarts in a future PR.  

To work around this issue, you could modify the example code above to create a new `SiteInst` based on the original cell's `SiteInst` type:

```java
public static void main(String[] args) throws FileNotFoundException {
    Design design1 = Design.readCheckpoint("add_placed.dcp");
    Design design2 = new Design("new_design", design1.getDevice().getName());
    for (Cell c : design1.getCells()) {
        // Manually add a new SiteInst here:
        SiteInst orig = c.getSiteInst();
        design2.createSiteInst(orig.getName(), orig.getSiteTypeEnum(), orig.getSite());
        design2.copyCell(c, c.getName());
    }
  }
```

Hope that helps and please follow up if there are other related issues.  Thanks!

nqdtan, 2022-05-13:
Thanks Chris for the detailed explanation! Your workaround fixes my issue. Resolved.

closed 2022-05-13 by nqdtan