Xilinx/RapidWright #422: [copyCell] Cells with hierarchical names return null EDIFCellInst

nqdtan, 2022-05-14:
(This is also related to the *copyCell* issue reported in #419. But I think the cause is quite different so I decided to file a new one instead.)

I have a design in which there are some cells with specific hierarchical names that crashes the tool when I tried to do copyCell(). I have created a minimal example that helps reproduce the issue.

**Verilog design**
```Verilog
module lut (
  input i0,                                                                      
  input i1,                                                                      
  input i2,                                                                      
  input i3,                                                                      
  output out0,                                                                   
  output out1                                                                    
);                                                                               
                                                                                 
  LUT2 #(                                                                        
    .INIT(4'h8)                                                                  
  ) i_ (                                                                         
    .I0(i0),                                                                     
    .I1(i1),                                                                     
    .O(out0)                                                                     
  );                                                                             
                                                                                 
  LUT2 #(                                                                        
    .INIT(4'h8)                                                                  
  ) \i_/a  (                                                                                                                                                                           
    .I0(i2),                                                                     
    .I1(i3),                                                                     
    .O(out1)                                                                     
  );                                                                             
endmodule
```
**RapidWright code**

```Java
public class TestCopyCell {                                                      
  public static void main(String[] args) throws FileNotFoundException {          
    Design design1 = Design.readCheckpoint("lut.dcp");               
    Design design2 = new Design("new_design", design1.getDevice().getName());       
    for (Cell c : design1.getCells()) {                                          
      System.out.println("Cell " + c.getName() + " " + c.getBEL() + " " + c.getSiteInst() + " " + c.getEDIFCellInst());
      SiteInst orig = c.getSiteInst();                                           
      design2.createSiteInst(orig.getName(), orig.getSiteTypeEnum(), orig.getSite());
      design2.copyCell(c, c.getName());                                          
    }                                                                            
  }                                                                              
}
```

The tool crashes when I have two LUTs with their instance names as above. The error that I got is

> Cell i_ H6LUT(BEL) SiteInst(name="SLICE_X182Y139", type="SLICEL", site="SLICE_X182Y139") i_
Cell i_/a G6LUT(BEL) SiteInst(name="SLICE_X182Y139", type="SLICEL", site="SLICE_X182Y139") null
Exception in thread "main" java.lang.NullPointerException
	at com.xilinx.rapidwright.design.Design.copyCell(Unknown Source)
	at com.xilinx.rapidwright.examples.TestCopyCell.main(TestCopyCell.java:18)

Since it fails to get an *EDIFCellInst* for cell *i_/a*, I suspect that might cause a crash in `copyCell`.

If I rename the second LUT from *\i_/a* to something like *a_*, there is no issue. Or if the design only contains one of the two LUTs, there's also no problem too.

Please find the source code in the attachment. I test this with RW 2022.1.1 and Vivado 2022.1.

Thank you!

[src.zip](https://github.com/Xilinx/RapidWright/files/8691857/src.zip)

clavin-xlnx, 2022-05-16:
I've created a PR with a fix for now.  I'm not certain about how we resolve this long term as I think this has runtime implications that would impact other flows.   But you can try it for your existing design (#426).

nqdtan, 2022-05-16:
Thanks Chris! Your PR works well for my design. One thing I'd like to bring up is that, if I do

```Java
for (Cell c : design1.getCells())
  System.out.println(c.getParentHierarchicalInstName());
```

for the example design above, I got the following exception

> Exception in thread "main" java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1967)
	at com.xilinx.rapidwright.design.Cell.getParentHierarchicalInstName(Unknown Source)
	at com.xilinx.rapidwright.examples.TestCopyCell.main(TestCopyCell.java:35)

I'll close the issue now as my problem is fixed.

closed 2022-05-16 by nqdtan

clavin-xlnx, 2022-05-17:
@nqdtan - Thanks for the heads up on the other issue.  To work around it (using #426), you can instead call:
```java
System.out.println(design1.getNetlist().getHierCellInstFromName(c.getName()).getParent().getFullHierarchicalInstName());
```

commit 5ef419f4 (2022-09-17): Merge pull request #426 from Xilinx/fix_422

Fix #422: Correctly resolve HierObject search with prefix match and '/' in name