Xilinx/RapidWright #464: Issues converting a post-placement .dcp file to an IF file

zhilix, 2022-06-28:
I ran this round-test command for physical netlist conversion:
`java com.xilinx.rapidwright.interchange.PhysicalNetlistExample FPGA01_postplace.dcp FPGA01_postplace_interchange.dcp`

And the errors messages are:
`zhilix@eda15:/home/local/eda15/zhilix/projects/RapidWright$ java com.xilinx.rapidwright.interchange.PhysicalNetlistExample FPGA01_postplace.dcp FPGA01_postplace_interchange.dcp
Exception in thread "main" java.lang.NullPointerException
        at com.xilinx.rapidwright.interchange.Enumerator.getKey(Enumerator.java:41)
        at com.xilinx.rapidwright.interchange.Enumerator.getIndex(Enumerator.java:52)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writeRouteBranch(PhysNetlistWriter.java:432)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writeRouteBranch(PhysNetlistWriter.java:470)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writeRouteBranch(PhysNetlistWriter.java:470)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.populateRouting(PhysNetlistWriter.java:413)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writePhysNets(PhysNetlistWriter.java:328)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writePhysNetlist(PhysNetlistWriter.java:505)
        at com.xilinx.rapidwright.interchange.PhysicalNetlistExample.main(PhysicalNetlistExample.java:67)`

Also the .dcp and .edf files dumped by vivado after I ran place_design are here:
[FPGA01_postplace.zip](https://github.com/Xilinx/RapidWright/files/9004305/FPGA01_postplace.zip)

clavin-xlnx, 2022-06-29:
Thank you @zhilix for the test example and error message.  It turns out this is stumbling on a known issue in the Interchange schema.  I have created a PR (#466) for you which you can use as a work-around until we get the schema updated and RapidWright changed to accommodate.  Please give it a try and see if you are able to convert your design.  As noted in the PR, you'll have to run `update_clock_routing` after loading design back into Vivado if you hope to route it.

zhilix, 2022-06-29:
Thanks! I test on that benchmark and I'm able to generate the IF files using .dcp after placement now!

clavin-xlnx, 2022-07-13:
@zhilix - This should be fixed now so that you no longer should need the work around.

PR #467 (2022-07-13): Changes to support placed (but not routed) designs in interchange
This depends on chipsalliance/fpga-interchange-schema#78 and fixes #464 and #136.  It also adds a git submodule for the Interchange schema at `RapidWright/interchange/fpga-interchange-schema`.