Xilinx/RapidWright #482: Read and write checkpoint and import to vivado2022.1 causing error

zhilix, 2022-07-26:
I used RapidWright's readCheckpoint and writeCheckpoint to convert ISPD16's FPGA05 and FPGA09 benchmarks' old DCP  to newer version of DCP.
But when I run open_checkpoint on vivado2022.1 to open the new DCP files, I got error messages like:
```
ERROR: [Designutils 20-1414] Failed to read XDEF archive '/home/local/eda15/zhilix/projects/RapidWright/rapidwright_dcp/FPGA05_rw_new/design80_1_inst0.xdef': input stream error. The most common reason for this failure is that the input file is corrupted or truncated. Make sure your storage device has sufficient free space to unzip the archive file. When this message is issued during a read_checkpoint operation, the netlist and constraints will reflect the contents of the checkpoint even though the place and route data was not read. Close the current design before attempting to retry the checkpoint read.
```
I have tried on different machines, still have the same error.

zhilix, 2022-07-26:
Here are the post-processed FPGA05 and FPGA09 DCPs:
[https://drive.google.com/drive/folders/1J4SV96pTJBGR-WXZjdN7nEeIlIlmjFMn](url)

clavin-xlnx, 2022-07-26:
Thanks @zhilix, although I am able to open the link, I am unable to download (I can see the files, but download fails).   Could you double check the files are correct?

Also, is it possible to get the pre-processed DCPs also (those DCPs you load into RapidWright before the conversion)?

zhilix, 2022-07-26:
Hi Chris, I updated with a new link:
 FPGA05_09.zip
<https://drive.google.com/file/d/1UIdZDD_eh4wb5bOpv14BBKyD6iN9_cn0/view?usp=drive_web>

clavin-xlnx, 2022-08-09:
@zhilix - There should be a fix for this issue in #491, give it a try and let me know if it works for you.  Thanks!

zhilix, 2022-08-10:
I checked out the #491, however I ran into error when I did the example tests:
```
zhilix@eda15:/home/local/eda15/zhilix/projects/RapidWright$ java com.xilinx.rapidwright.interchange.PhysicalNetlistExample FPGA01_handplace.dcp FPGA01_handplace.edf FPGA01_handplace_new.dcp
Exception in thread "main" java.lang.NoClassDefFoundError: com/xilinx/rapidwright/design/Design
        at com.xilinx.rapidwright.interchange.PhysicalNetlistExample.main(PhysicalNetlistExample.java:57)
Caused by: java.lang.ClassNotFoundException: com.xilinx.rapidwright.design.Design
        at java.net.URLClassLoader.findClass(URLClassLoader.java:387)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:418)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:352)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:351)
        ... 1 more
```

clavin-xlnx, 2022-08-10:
Sorry, you'll need to update the jar when switching to a different release run `./gradlew update_jars` to update the jar.  You might also need to update the CLASSPATH, or you can specify it directly on the command line:
```
java -cp bin:jars/* com.xilinx.rapidwright.interchange.PhysicalNetlistExample FPGA01_handplace.dcp FPGA01_handplace.edf FPGA01_handplace_new.dcp
```

zhilix, 2022-08-11:
Yes, this works, thank you very much!

clavin-xlnx, 2022-08-25:
Closing as it appears fixed.

closed 2022-08-25 by clavin-xlnx