Xilinx/RapidWright #494: PhysicalNetlistToDcp converting .phys(only placements) to DCP

zhilix, 2022-08-05:
I tried to use PhysicalNetlistToDcp  to convert my placments in IF to DCP, and I commented off the xdc input requirement.
```
zhilix@eda15:/home/local/eda15/zhilix/projects/RapidWright$ java com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp FPGA01_rapidwright.netlist example_FPGA01_placement.phys example_FPGA01_rapidwright.dcp
Exception in thread "main" java.lang.NullPointerException
        at com.xilinx.rapidwright.interchange.PhysNetlistReader.readPlacement(PhysNetlistReader.java:216)
        at com.xilinx.rapidwright.interchange.PhysNetlistReader.readPhysNetlist(PhysNetlistReader.java:111)
        at com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp.main(PhysicalNetlistToDcp.java:60)
```
It seems that Rapidwright can't read physical netlist that only has placments field.

Here are the .netlist and .phys files that I used:
[netlist_phys.zip](https://github.com/Xilinx/RapidWright/files/9269966/netlist_phys.zip)
.

clavin-xlnx, 2022-08-09:
I was able to reproduce the issue.  Please try again with #495, however, I think there is information still missing that is not being populated.  I have changed RapidWright to be more forgiving, however, the file is missing the `SiteInst` information, those fields are empty.  

Also, when I attempted to load the design into Vivado, it crashed, so there is something else that is missing.  Unfortunately, I don't know what that might be but if you could start with an equivalent file in Vivado of what you think you should expect, then convert that back to Interchange, that should give you a hint of information you might be missing.

zhilix, 2022-08-09:
Thanks a lot, I think I'll start adding other fields as well, it seems other fields are also necessary for Vivado to load the design successfully.

eddieh-xlnx, 2022-08-25:
Did #495 fix this? If so, we can close.

zhilix, 2022-08-25:
Yes, this is fixed, please close it.

closed 2022-08-25 by clavin-xlnx