Xilinx/RapidWright #504: Ran routeSite() causing NullPointerException

zhilix, 2022-08-18:
I made some changes to the PhysicalNetlistToDcp.java, just to test the intra-site routing:
```
package com.xilinx.rapidwright.interchange;

import java.io.File;
import java.io.IOException;
import java.nio.charset.Charset;
import java.nio.file.Files;
import java.util.List;
import java.util.*;

import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.SiteInst;
import com.xilinx.rapidwright.edif.EDIFNetlist;
import com.xilinx.rapidwright.tests.CodePerfTracker;
import com.xilinx.rapidwright.design.ConstraintGroup;

public class PhysicalNetlistToDcp {

    public static void main(String[] args) throws IOException {
        if(args.length != 3) {
            System.out.println("USAGE: <input>.netlist <input>.phys <input>.xdc <output>.dcp");
            System.exit(1);
            return;
        }

        String logNetlistFileName = args[0];
        String physNetlistFileName = args[1];
        // String xdcFileName = args[2];
        String outputDCPFileName = args[2];

        CodePerfTracker t = new CodePerfTracker("Interchange Format->DCP",false);

        t.start("Read Logical Netlist");

        // Read Netlist into RapidWright netlist
        EDIFNetlist n2 = LogNetlistReader.readLogNetlist(logNetlistFileName);

        t.stop().start("Read Physical Netlist");

        // Read Physical Netlist into RapidWright netlist
        Design roundtrip = PhysNetlistReader.readPhysNetlist(physNetlistFileName, n2);

        // Collection<SiteInst> site_insts = roundtrip.getSiteInsts();

        // for (SiteInst instance : site_insts) {
        //     instance.routeSite();
        // }

        roundtrip.routeSites();

        // Add XDC constraints
        // List<String> lines = Files.readAllLines(new File(xdcFileName).toPath(), Charset.defaultCharset());
        // roundtrip.setXDCConstraints(lines, ConstraintGroup.NORMAL);

        t.stop().start("Write DCP");

        // Write RapidWright netlist back to edif
        roundtrip.writeCheckpoint(outputDCPFileName, CodePerfTracker.SILENT);

        t.stop().printSummary();
    }
}
```

And then I ran command like this:
```
zhilix@eda15:/home/local/eda15/zhilix/projects/RapidWright$ java com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp FPGA01_rapidwright.netlist example_FPGA01_ports.phys example_FPGA01_ports.dcp
Exception in thread "main" java.lang.NullPointerException
        at com.xilinx.rapidwright.design.SiteInst.routeSite(Unknown Source)
        at com.xilinx.rapidwright.design.Design.routeSites(Unknown Source)
        at com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp.main(PhysicalNetlistToDcp.java:70)
```
The file used:
[netlist&phys.zip](https://github.com/Xilinx/RapidWright/files/9376442/netlist.phys.zip)

clavin-xlnx, 2022-08-19:
Looking at this example, the reason for the failure is that the pin mappings on the input Interchange design are not correct.  I've added a warning message that should help you resolve the mismatch in #506.  For example, here are some of the issues that I saw:
```
WARNING: On cell inst_13935, a logical pin 'LUT4.I3' is being mapped on to a BEL pin 'G5LUT.A6' that does not exist.
WARNING: On cell inst_13935, a logical pin 'LUT4.O' is being mapped on to a BEL pin 'G5LUT.O6' that does not exist.
```
What this is saying is that you have a `LUT4` being placed onto the BEL location `G5LUT`, however a `G5LUT` does not have an `A6` pin nor does it have an `O6` pin:

[image]

It is possible that you meant to place it on the `G6LUT` which does have those pins, but you'll have to look at your intent upstream to be sure.

zhilix, 2022-08-22:
Thanks! I was able to eliminate the pin mapping warnings. However, I still meet another error.
```
zhilix@eda15:/home/local/eda15/zhilix/projects/RapidWright$ java com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp FPGA01_rapidwright.netlist example_FPGA01_properties.phys example_FPGA01_properties.dcp
Exception in thread "main" java.lang.RuntimeException: ERROR: Couldn't route net net_69836 in site SLICE_X57Y127
        at com.xilinx.rapidwright.design.SiteInst.a(Unknown Source)
        at com.xilinx.rapidwright.design.SiteInst.routeSite(Unknown Source)
        at com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp.main(PhysicalNetlistToDcp.java:67)
```
And I wonder where this routesite() source java code is and I probably can reference to it.

clavin-xlnx, 2022-08-22:
> Thanks! I was able to eliminate the pin mapping warnings. However, I still meet another error.
>
> ```
> zhilix@eda15:/home/local/eda15/zhilix/projects/RapidWright$ java [...]
Could you share the updated design input files?  I could try to debug the issue now that they have changed, I don't think I can reproduce the issue.

> And I wonder where this routesite() source java code is and I probably can reference to it.
This code is actually in the closed source portion of RapidWright, so it is currently not accessible.

zhilix, 2022-08-23:
Yes, here are the updated files:
[new_phys_netlist.zip](https://github.com/Xilinx/RapidWright/files/9398327/new_phys_netlist.zip)

clavin-xlnx, 2022-08-23:
It looks like there are pin mapping overlaps on a LUT.  Adding this code after reading the physical netlist:
```java
        // Read Physical Netlist into RapidWright netlist
        Design roundtrip = PhysNetlistReader.readPhysNetlist(physNetlistFileName, n2);

        for(Cell cell : roundtrip.getSiteInst("SLICE_X57Y127").getCells()) {
            if(cell.getBELName().startsWith("H") && cell.getBELName().endsWith("LUT")) {
                for(Entry<String,String> pin : cell.getPinMappingsP2L().entrySet()) {
                    String belPinName = pin.getKey();
                    String logicalCellPinName = pin.getValue();
                    System.out.println(cell.getSiteName() + "/" + cell.getBELName() + " " 
                            + belPinName + ":" + logicalCellPinName + " " 
                            + "net=" + cell.getEDIFHierCellInst().getPortInst(logicalCellPinName).getHierarchicalNet() + " "
                            + "cell=" + cell.getName());
                }
            }
        }
```

yields:
```
SLICE_X57Y127/H6LUT A4:I0 net=net_72891 cell=inst_72943
SLICE_X57Y127/H6LUT A5:I1 net=net_72889 cell=inst_72943
SLICE_X57Y127/H6LUT A6:I2 net=net_72888 cell=inst_72943
SLICE_X57Y127/H6LUT O6:O net=net_72890 cell=inst_72943
SLICE_X57Y127/H5LUT A4:I0 net=net_69839 cell=inst_69891
SLICE_X57Y127/H5LUT A5:I1 net=net_69836 cell=inst_69891
SLICE_X57Y127/H5LUT O5:O net=net_69838 cell=inst_69891
```
So H5LUT and H6LUT share their inputs and two different cells are being placed onto those locations `(inst_69891 -> SLICE_X57Y127/H5LUT and inst_72943 -> SLICE_X57Y127/H6LUT)` but their inputs are not driven by the same net.  In this case the placement dictates two different nets should be routed to the same pin (`A5`) which is not possible. 

Another issue is that when you use both the `5LUT` and `6LUT` at the same time, the `A6` pin needs to be driven by VCC so the pin becomes unusable and the pins must be mapped to one of the other pins.  However, it is good to start with the higher numbered pins first and work backwards as the higher number pins have lower delay.

zhilix, 2022-08-26:
Hi, it seems I'm keeping meeting new bugs when I'm running this

zhilix, 2022-08-26:
It seems I'm keeping meeting bugs

zhilix, 2022-08-26:
This time I fixed the overlap, but still meet another NullPointerException
```
zhilix@eda15:/home/local/eda15/zhilix/projects/RapidWright$ java com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp FPGA01_rapidwright.netlist example_FPGA01_no_overlap.phys example_FPGA01_routesite.dcp
Exception in thread "main" java.lang.NullPointerException
        at com.xilinx.rapidwright.design.Cell.getEDIFCellInst(Unknown Source)
        at com.xilinx.rapidwright.design.SiteInst.routeSite(Unknown Source)
        at com.xilinx.rapidwright.design.Design.routeSites(Unknown Source)
        at com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp.main(PhysicalNetlistToDcp.java:70)
```
[no_overlap.zip](https://github.com/Xilinx/RapidWright/files/9435773/no_overlap.zip)

clavin-xlnx, 2022-08-26:
@zhilix - I've diagnosed the issue and created a fix, in order to get it you'll have to use the `2022.1.3` branch.  From the command line, you could do the following to get the update:
```
git fetch
git checkout 2022.1.3
./gradlew updateJars
./gradlew compileJava
```

zhilix, 2022-09-01:
Just to give an update about this issue, I still got issues with IOs on vivado by running "route_design".
I think I got the A6 pin problem fixed in my end. and there are still critical warnings like:
```
CRITICAL WARNING: [Route 35-19] Pin mapping failure, cannot reach driver pin: inst_79/IBUFCTRL_INST/O at site IOB_X1Y183. Design will fail DRC and router will skip routing of net inst_79/O.
Resolution: For technical support on this issue, please visit http://www.xilinx.com/support
```
The physical and logical netlist files:
[phys_logical.zip](https://github.com/Xilinx/RapidWright/files/9472306/phys_logical.zip)

The DCP I got after running routeSites(), and the log file from vivado:
[dcp_log.zip](https://github.com/Xilinx/RapidWright/files/9472310/dcp_log.zip)

clavin-xlnx, 2022-09-08:
@zhilix - With the input files in `phys_logical.zip`, I did not get the `CRITICAL WARNING` you cited above with the latest changes on `2022.1.3`.  Perhaps you could double check or perform a clean checkout of your RapidWright project?

Routing proceeds normally and ends successfully.  Although, there do seem to be some missing VCC connections on dual-used LUTs.  When I run `report_route_status`:
```
report_route_status
Design Route Status
                                               :      # nets :
   ------------------------------------------- : ----------- :
   # of logical nets.......................... :      105481 :
       # of nets not needing routing.......... :        7147 :
           # of internally routed nets........ :        7147 :
       # of routable nets..................... :       98334 :
           # of fully routed nets............. :       98333 :
       # of nets with routing errors.......... :           1 :
           # of nets with some unrouted pins.. :           1 :
   ------------------------------------------- : ----------- :

Nets with Routing Errors:
  GLOBAL_LOGIC1
    Unrouted Nodes (not associated with pins or ports) -- only the first 10 are listed, use -show_all to get the full list:
      INT_X24Y117/IMUX_E16
      INT_X24Y119/IMUX_E16
      INT_X24Y119/IMUX_E17
      INT_X24Y120/IMUX_W16
      INT_X24Y120/IMUX_W46
      INT_X24Y121/IMUX_E28
      INT_X25Y114/IMUX_E16
      INT_X25Y114/IMUX_E34
      INT_X25Y115/IMUX_E28
      INT_X25Y116/IMUX_E16
```
Those unrouted nodes are supposed to be have VCC driving LUT A6 inputs:

[image]

Otherwise, it looks good.

zhilix, 2022-09-09:
Thanks a lot!! I actually cleaned the checkout and added the Vcc connections to the site pin A6 inputs.
Now I got 0 error after running "route_design -no_timing_driven" and "report_route_status".

clavin-xlnx, 2022-10-21:
Closing for now.

closed 2022-10-21 by clavin-xlnx