Xilinx/RapidWright #512: Some libraries are not copied

pongstorn-xilinx, 2022-08-25:
EDIFNetlist.copyCellAndSubCells fails to copy some libraries associated with the given cell.

eddieh-xlnx, 2022-08-25:
Please provide a standalone testcase, one which can serve as a test for the future fix.

eddieh-xlnx, 2022-08-25:
It looks like `copyCellAndSubCells()` is copying cells into the work library:
https://github.com/Xilinx/RapidWright/blob/c2dad6174883913410ec9cdce6c7884f0d0e52ab/src/com/xilinx/rapidwright/edif/EDIFNetlist.java#L550-L556

Since `copyCellAndSubCells()` grew out from `migrateCellAndSubCells`:
https://github.com/Xilinx/RapidWright/blob/c2dad6174883913410ec9cdce6c7884f0d0e52ab/src/com/xilinx/rapidwright/edif/EDIFNetlist.java#L440-L446

@clavin-xlnx Is there any reason that I'm missing about why new cells and dumped into the work library rather than creating a new library with the same name, should it not exist?

clavin-xlnx, 2022-08-25:
> @clavin-xlnx Is there any reason that I'm missing about why new cells and dumped into the work library rather than creating a new library with the same name, should it not exist?

Nope.  Dumping all the cells into work was the defacto solution in the past, but we've grown robust enough that we should be able to propagate all libraries going forward.

PR #513 (2022-08-25): EDIFNetlist.{copy,migrate}CellAndSubCells() to copy into new same-named library
Rather than into the work library, if source library doesn't exist in destination netlist.

Fixes #512.
closed 2022-08-25 by eddieh-xlnx