Xilinx/RapidWright #546: RapidWrigth update: RuntimeException: Expected routethru cell

robertszczepanski, 2022-09-27:
Hi, there is an issue with updating RapidWright in [chipsalliance/fpga-tool-perf](https://github.com/chipsalliance/fpga-tool-perf) repository. I was able to bisect this issue to 3042468 commit which reverts some stuff, does it require any adjustments on our side?

Error message:
```
source /hdd/fpga-tool-perf/env.sh nextpnr fpga_interchange-a100t nextpnr && RAPIDWRIGHT_PATH=/hdd/fpga-tool-perf/third_party/RapidWright /hdd/fpga-tool-perf/third_party/RapidWright/scripts/invoke_rapidwright.sh com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp attosoc.netlist attosoc.phys /hdd/fpga-tool-perf/src/attosoc/constr/arty-a100t.xdc attosoc.dcp
Exception in thread \"main\" java.lang.RuntimeException: Expected routethru cell at SLICE_X75Y128/A6LUT
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:452)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouteBranch(PhysNetlistReader.java:501)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readRouting(PhysNetlistReader.java:366)
   at com.xilinx.rapidwright.interchange.PhysNetlistReader.readPhysNetlist(PhysNetlistReader.java:115)
   at com.xilinx.rapidwright.interchange.PhysicalNetlistToDcp.main(PhysicalNetlistToDcp.java:60)
make: *** [Makefile:23: attosoc.dcp] Error 1
```

Steps to reproduce:
```
git clone https://github.com/chipsalliance/fpga-tool-perf && cd fpga-tool-perf
git submodule update --init --recursive 
cd third_party/RapidWright && git checkout 3042468 && cd ../..
make install_interchange
TOOLCHAIN=nextpnr make env
source env.sh nextpnr && python3 exhaust.py --project attosoc --toolchain nextpnr-fpga-interchange --board arty-a35t --build_type attosoc-nextpnr-fpga-interchange-arty-a35t --fail --timeout 7200 --verbose
```

eddieh-xlnx, 2022-09-28:
Can you attach the logical and physical Interchange netlists please?

robertszczepanski, 2022-09-29:
[netlists.zip](https://github.com/Xilinx/RapidWright/files/9672277/netlists.zip)

Sure, I generated these with:
```
python3 exhaust.py --project vexriscv --toolchain nextpnr-fpga-interchange --board arty-a100t --build_type vexriscv-nextpnr-fpga-interchange-arty-a100t --fail --timeout 7200
```

eddieh-xlnx, 2022-10-03:
Thanks for reporting this! Looks like there was a change of behaviour in #486 (from which https://github.com/Xilinx/RapidWright/commit/30424682618c64533fffdedeb05d9e58159571e9 is derived) that silently broke how LUT routethrus could be inferred. This should be fixed in #554. Please check the read design is what you expect.

robertszczepanski, 2022-10-05:
Thanks @eddieh-xlnx, I just had a chance to test this on hardware and I can confirm that with your fix I am able to build working bitstream.

PR #554 (2022-10-05): [Interchange] Create LUT routethru cells if not present
Fixes #546
closed 2022-10-05 by eddieh-xlnx