Xilinx/RapidWright #580: Provide a routed net helper method that automatically sets PIP.setIsReversed() correctly for used bidirectional PIPs

clavin-xlnx, 2022-11-10:
For #575 and future updates to work correctly, PIPs will need to properly set `PIP.setIsReversed()` on bidirectional PIPs.  This is most optimally done at the point of creation in the correct context.  This issue requests a new helper method that can be used to ensure and/or correct PIPs that do not have this flag set properly with the expense of O(n), where n is the number of PIPs on the net.

eddieh-xlnx, 2023-08-03:
@clavin-xlnx In light of #774, is there still a need for this helper method?

This issue references #575, which is responsible for unrouting nets, stating that it relies on `PIP.isReversed()` to be correct to function correctly. With #774, RWRoute should now set this flag correctly.

clavin-xlnx, 2023-08-03:
> @clavin-xlnx In light of #774, is there still a need for this helper method?
> 
> This issue references #575, which is responsible for unrouting nets, stating that it relies on `PIP.isReversed()` to be correct to function correctly. With #774, RWRoute should now set this flag correctly.

I suppose it becomes less important, I suppose for other users developing routing solutions they can either manage the issue on their own or write their own helper method.

eddieh-xlnx, 2025-05-13:
Closing as this appears resolved.

closed 2025-05-13 by eddieh-xlnx