Xilinx/RapidWright #607: How to access BEL Properties?

jgoeders, 2023-01-25:
Perhaps I am not seeing it in the documentation, but is there a way to access BEL Properties?  

For example, in Vivado I can look at this A6LUT CONFIG.EQN property, as well as access it via Tcl:

```
get_property CONFIG.EQN [get_bels SLICE_X0Y102/A6LUT]
O6=(A3*A4*A5)
```

[image]

My understanding is that in RapidWright, the BEL class has no design/configuration information, so I would expect this in the Cell class (since there is no BELInst)?  But retrieving properties from the Cell class only provides the EDIF properties.  For example, I can get the INIT string from the logical netlist, but not the CONFIG.EQN for the physical LUT implementation.

....my ultimate goal is to get the physical INIT string for the BEL.  I don't see this property in Vivado, so my current approach is to get the CONFIG.EQN and build the INIT string (possibly using the [LUTTools](http://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/tools/LUTTools.html).

Am I missing something, or is this information not available in RapidWright?

Thanks!

clavin-xlnx, 2023-01-25:
Thanks for bringing up this issue.  It appears that the `CONFIG.EQN` property on BELs is not something that is stored directly in the DCP and I'm assuming it would be re-generated upon load of the design.  Currently, RapidWright doesn't calculate/re-generate these values on BELs when loading a DCP, but is theoretically possible.  

As you mentioned, the `LUTTools` class has some utility methods around converting the Cell `INIT` string into a LUT equation, but the result will be different in two ways.  First, the method `LUTTools.getLUTEquation(Cell)` looks at the `INIT` property and generates a very verbose LUT equation (essentially reading off the truth table).  The second way it is different is that it uses logical inputs (I0, I1, I2, ...) whereas, the `CONFIG.EQN` property uses the physical BEL pins (A1, A2, A3, ...).  

To resolve the first difference, a logic minimization technique would need to be employed (something like [Quine-McCluskey](https://en.wikipedia.org/wiki/Quine%E2%80%93McCluskey_algorithm)).  RapidWright currently doesn't have this implemented.  However, although this would significantly reduce the length of the equation string, it wouldn't always necessarily guarantee the same string as what appears in Vivado depending on how the inputs are arranged, etc.

To resolve the second difference, the logical pin names would need to be replaced by the physical pin name mappings which are stored on the `Cell` object.  The other thing that would be missing is if both LUTs (5LUT and 6LUT) are being used in a BEL, the A6 input must be VCC and this is expressed in the `CONFIG.EQN` as `(A6+~A6)`, which would need to be suffixed (ANDed) to the equation when both LUTs are being used. Here is some sample code to illustrate this second part:

```java
        // this is from the test/RapidWrightDCP test suite, 
        // wget https://github.com/eddieh-xlnx/RapidWrightDCP/raw/master/bnn.dcp
        Design design = Design.readCheckpoint("bnn.dcp");
        Cell c = design.getCell("bd_0_i/hls_inst/inst/grp_fp_conv_fu_507/n_V_reg_2113[1]_i_1");
        
        String lutEquation = LUTTools.getLUTEquation(c);
        System.out.println("Logical LUT EQ = " + lutEquation);
        for(String logicalPinName : c.getPinMappingsL2P().keySet()) {
            String physPinName = c.getPhysicalPinMapping(logicalPinName);
            lutEquation = lutEquation.replace(logicalPinName, physPinName);
        }
        // If the other LUT BEL is being used, add A6 nop string
        if (c.getSiteInst().getCell(LUTTools.getCompanionLUTName(c.getBEL())) != null) {
            lutEquation = lutEquation.replace("=", "=(A6 + ~A6)*");
        }
        System.out.println("Physical LUT EQ = " + lutEquation);
```

The relevant output of this code is:
```
Logical LUT EQ = O=!I1 & I0 + I1 & !I0
Physical LUT EQ = O6=(A6 + ~A6)*!A2 & A5 + A2 & !A5
```
In the Vivado cell view:
[image]

In the Vivado BEL view:
[image]

jgoeders, 2023-02-20:
Thanks, this has worked well!

closed 2023-02-20 by jgoeders