Xilinx/RapidWright #621: RWRoute routes to additional, unnecessary pins

clavin-xlnx, 2023-03-09:
To reproduce:
[example.zip](https://github.com/Xilinx/RapidWright/files/10935988/example.zip)
```
unzip example.zip
java com.xilinx.rapidwright.rwroute.RWRoute placed.dcp output_routed.dcp --nonTimingDriven
```
When reading the result back into Vivado, `report_route_status` returns:
```
report_route_status
Design Route Status
                                               :      # nets :
   ------------------------------------------- : ----------- :
   # of logical nets.......................... :        4738 :
       # of nets not needing routing.......... :        1044 :
           # of internally routed nets........ :         835 :
           # of nets with no loads............ :          50 :
           # of implicitly routed ports....... :         159 :
       # of routable nets..................... :        3694 :
           # of unrouted nets................. :         106 :
           # of fully routed nets............. :        1979 :
       # of nets with routing errors.......... :        1664 :
           # of nets with some unrouted pins.. :         823 :
           # of nets with resource conflicts.. :        1579 :
           # of nets with antennas/islands.... :        1156 :
   ------------------------------------------- : ----------- :
```
Looking at a specific net (`ipr[15]`) with a conflict, it would appear that the router routed to additional sinks unnecessarily for reasons unknown:
[image]
[image]

clavin-xlnx, 2023-03-10:
Sources for the placed design in Interchange format
[interchange_src.zip](https://github.com/Xilinx/RapidWright/files/10938587/interchange_src.zip)

eddieh-xlnx, 2023-03-27:
Looks like the input DCP was corrupted (originally and incorrectly converted from the Interchange format, which [now](https://github.com/Xilinx/RapidWright/pull/627) detects this occurrence).

closed 2023-03-27 by eddieh-xlnx