Xilinx/RapidWright #630: RWRoute is throwing a runtime exception in `findReferenceTile()`

burntfalafel, 2023-03-19:
Title says it all. I am trying to use the RWRoute of RapidWright on a design (verilog file) of mine. 

I created the dcp in Vivado using the following commands: 
```
launch_runs synth_1 -jobs 8
open_run synth_1 -name synth_1
place_ports
source ../../tcl/rapidwright.tcl
write_rw_checkpoint test
```

I then do a `java com.xilinx.rapidwright.rwroute.RWRoute ./test.dcp routed.dcp` but the output is as follows:
```
==============================================================================
==                                 RWRoute                                  ==
==============================================================================
==============================================================================
==                          Reading DCP: test.dcp                           ==
==============================================================================
 XML Parse & Device Load:     0.421s
    (Read PhysDB Caches):            (    0.071s)
   (Read PhysDB Routing):            (    0.031s)
 (Read PhysDB Placement):            (    0.107s)
(Read PhysDB Finalizing):            (    0.000s)
            (EDIF Parse):            (    0.580s)
   Read Netlist & PhysDB:     0.582s
------------------------------------------------------------------------------
         [No GC] *Total*:     1.002s
WARNING: Not masking nodes across RCLK could result in delay optimism.
Exception in thread "main" java.lang.NullPointerException
        at com.xilinx.rapidwright.timing.TimingModel.findReferenceTile(TimingModel.java:477)
        at com.xilinx.rapidwright.timing.TimingModel.readDelayTerms(TimingModel.java:498)
        at com.xilinx.rapidwright.timing.TimingModel.build(TimingModel.java:243)
        at com.xilinx.rapidwright.timing.delayestimator.DelayEstimatorBase.<init>(DelayEstimatorBase.java:87)
        at com.xilinx.rapidwright.rwroute.RWRoute.createRouteNodeGraph(RWRoute.java:229)
        at com.xilinx.rapidwright.rwroute.RWRoute.initialize(RWRoute.java:184)
        at com.xilinx.rapidwright.rwroute.RWRoute.routeDesign(RWRoute.java:1753)
        at com.xilinx.rapidwright.rwroute.RWRoute.routeDesign(RWRoute.java:1741)
        at com.xilinx.rapidwright.rwroute.RWRoute.routeDesignWithUserDefinedArguments(RWRoute.java:1733)
        at com.xilinx.rapidwright.rwroute.RWRoute.main(RWRoute.java:1778)
```

Any ideas on how I can fix this? [Attached](https://file.io/2g3OiSTlYLDD) the `test.dcp` file for reference

clavin-xlnx, 2023-03-22:
@burntfalafel - thanks for filing this issue!  It turns out that RWRoute doesn't support Series 7 devices which the target example design is targeting (`xc7s100fgga676-1Q`).  Technically, RWRoute only supports UltraScale+ devices (although, some UltraScale designs work too).  Try again by targeting an UltraScale+ device and let us know if you still run into issues.

I've filed issue #632 to add an error message when attempting to route an unsupported device.

burntfalafel, 2023-03-23:
@clavin-xlnx thx, Is there any reason why series 7 devices are not supported?

I will close this now.

closed 2023-03-23 by burntfalafel

clavin-xlnx, 2023-03-27:
@burntfalafel - no technical reason that Series 7 devices are not supported.  Could easily be added with some effort, we just have limited resources to do so.