Xilinx/RapidWright #631: Cell placements missing for Versal Premium

gatecat, 2023-03-20:
I was looking at the possibility of RapidWright/nextpnr on this device (or more realistically regions thereof), however, my import script isn't working because there are no cell->bel placements.

```java
package net.gatecat.testing;

import com.xilinx.rapidwright.design.*;
import com.xilinx.rapidwright.device.*;
import com.xilinx.rapidwright.edif.*;
import java.io.IOException;
import java.util.*;

public class VersalIOPlacementTest {
    public static void main(String[] args) throws IOException {
        Design des = new Design("top", "xcvp1802-lsvc4072-2MP-e-S");
        Device dev = des.getDevice();
        for (EDIFCell prim : Design.getPrimitivesLibrary(dev.getName()).getCells()) {
            Cell c = des.createCell("inst", prim);
            System.out.println("Cell type: " + c.getType());
            for (Map.Entry<SiteTypeEnum, Set<String>> entry : c.getCompatiblePlacements().entrySet()) {
                System.out.println("  site type: " + entry.getKey().toString());
                for (String bel : entry.getValue()) {
                    System.out.println("    bel: " + bel);
                }
            }
            des.removeCell(c);
            des.getNetlist().getTopCell().removeCellInst("inst");
        }
    }
}

```

should print them, but only prints the list of primitives and not any placements at all.

clavin-xlnx, 2023-03-22:
Thanks @gatecat for bringing this up.  I am able to confirm that the compatible placements are missing.  It seems there is likely a bug in the generation of the data file that supplies this API.  I'll be in a better position to follow up and provide more information next week to help resolve this issue.

clavin-xlnx, 2023-03-30:
@gatecat - found that there was an aliasing problem for the Versal architectures, a new data file with the placement information on Versal Premium devices has been updated with #640.  Let us know if you run into any other issues.

eddieh-xlnx, 2023-05-02:
Assumed fixed by #640. Please re-open if not!

closed 2023-05-02 by eddieh-xlnx