Xilinx/RapidWright #649: [ABCD]OUTMUX not always showing up in siteInst.getUsedSitePIPs()

cfib, 2023-04-11:
If a signal is routed via CLBLM_R.CLBLM_L_A->>CLBLM_L_AMUX (or equivalent) this seems to actually be done using the slice's AOUTMUX:

[image]

[image]

However, when I subsequently do the following in RapidWright

    design = Design.readCheckpoint('mwe.dcp')
    for siteInst in design.getSiteInsts():
        for usp in siteInst.getUsedSitePIPs():
            print(siteInst, usp)

I only get

    SiteInst(name="SLICE_X39Y20", type="SLICEL", site="SLICE_X39Y20") AUSED.0->>OUT

but **not** the additionally expected AOUTMUX.O6->>OUT.

I verified that when I use A5LUT instead (and thus O5 is used), I get the expected entry for AOUTMUX:

    SiteInst(name="SLICE_X39Y20", type="SLICEL", site="SLICE_X39Y20") AOUTMUX.O5->>OUT

I am using Vivado 2019.1 and RapidWright 2022.2.2.

The DCP, EDF, and Python script for the first case (AOUTMUX.O6->>OUT) is attached here: [mwe.zip](https://github.com/Xilinx/RapidWright/files/11205145/mwe.zip)

cfib, 2023-04-11:
Just as an addition: 
If I do the following in the Vivado Tcl console:

    report_property [get_site_pips -of [get_sites SLICE_X39Y20] SLICE_X39Y20/AOUTMUX:O6]

I get the following output

    Property  Type    Read-only  Value
    CLASS     string  true       site_pip
    FROM_PIN  string  true       O6
    IS_FIXED  bool    true       0
    IS_USED   bool    true       1
    NAME      string  true       SLICE_X39Y20/AOUTMUX:O6
    SITE      string  true       SLICE_X39Y20
    TO_PIN    string  true       OUT

whereas 

    report_property [get_site_pips -of [get_sites SLICE_X39Y20] SLICE_X39Y20/AOUTMUX:O5]

reports

    Property  Type    Read-only  Value
    CLASS     string  true       site_pip
    FROM_PIN  string  true       O5
    IS_FIXED  bool    true       0
    IS_USED   bool    true       0
    NAME      string  true       SLICE_X39Y20/AOUTMUX:O5
    SITE      string  true       SLICE_X39Y20
    TO_PIN    string  true       OUT

The `IS_USED == 1` property in the first Tcl command output indicates to me that Vivado knows about this situation.

clavin-xlnx, 2023-04-12:
Thanks for bringing up this issue.  I took a look at your example and indeed Vivado marks the BELPin on `SLICE_X39Y20/AOUTMUX:O6` as used.  However, when the design is written out as a DCP, the site pip is not stored in the file, likely due to an inconsistency in Vivado.   Vivado is able to reconstitute the site routing correctly, but RapidWright has not been instrumented to handle this inconsistency.  

Can you share more about what you might be trying to accomplish?  Perhaps there is way we can help.  If you are trying to achieve a specific routing, that is certainly possible with RapidWright.

cfib, 2023-04-12:
Thanks for the quick reply!
I am just trying to export the mapping between routing resources and nets for the analysis of fault injection results. So, for now, I can just work around this issue by treating the PIP CLBxxxxx_[ABCD]->>CLBxxxxx_[ABCD]MUX as a special case.

clavin-xlnx, 2023-04-13:
Thank you for reporting the issue, it is good that we are aware of it.  

Let us know if you have any trouble creating a work around for this inconsistency.

eddieh-xlnx, 2023-05-02:
Closing as workaround exists.

closed 2023-05-02 by eddieh-xlnx